MarlinFirmware/Marlin #7576: G30 command: Did anything change between 1.1.3 to 1.1.4 regarding the reported height?

pandel, 2017-08-27:
Hi all!

Sorry for the somewhat strange title, but I didn't know how to ask better... I'll try to explain my problem.

Last weekend I updated my firmware from old Marlin 1.1.3 based Skynet 2.3.2 to current Marlin bugfix-1.1.x branch, because I wanted to try out some of the new features.

Now I noticed some kind of strange behaviour regarding the G30 command. Up to now I used to setup my Z probe offset like this (I'm using a IR sensor):

(LINEAR leveling, 3x3)
- heat up bed and nozzle
- M851 Z0
- M500
- G28
- G1 X110 Y110 (center of bed)
- nozzle down until it grabs a piece of paper
- G92 Z0
- **G30** X110 Y110
  -> this always returned something like Z:1.30 or Z:1.45, which was nearly the exact difference between the tip of the nozzle and the probe trigger point
- M851 Z-<Z position value from last command)
- M500

@oderwat suggested this method some time ago (see https://github.com/MarlinFirmware/Marlin/issues/4612#issuecomment-279731674)and it always worked reliably and gave a very good base for the  following G29.

Even though my prints where ok so far, I thought that the first layer might get even better with BILINEAR leveling - my bed is not the best :)... So I set up the new firmware with BILINEAR leveling and a 4x4 grid, flashed the firmware, heated up the nozzle and the bed, cleaned the nozzle's outside a bit and after that I tried to determine the Z probe offset, just to be sure, with the method described above....

**BUT now,** when I use the same procedure as listed above, the G30 command seems to only report some kind of measurement deviation like 0.01 or 0.02, but not the probe offset position like before. So at the moment, I can only determine the probe offset reading the value on the printer's display, which isn't as accurate as G30 was in the former firmware version and it's a lot more fiddling and tweaking to get it right.

I noticed in the source code that some things have changed in probe_pt and run_z_probe so this might be the reason but I'm not deep enough into the source that I could correctly interpret what is going on.

Please, can someone tell me what's going on here? Or is there another good way to determine the probe offset which I may have not noticed yet?

Cheers, Holger

pandel, 2017-08-29:
Thanks empakoso!

I read this before, but I wouldn't have said that this is especially Anet related, but more a basic change from 1.1.3 to 1.1.4 somehow. Neither did I have any problems regarding the bootloader nor did I have firmware size issues. Superficially, it is just the changed behaviour of G30 that I don't understand.

benlye, 2017-08-29:
@pandel - I'm not at my printer, but I vaguely recall observing the same thing.  Assuming that `G30` has changed for some reason, instead of `G92 Z0` and `G30` you could try using `M114` to give you the current Z position when the nozzle is gripping the paper and set that using `M851`.  Just do `M114` where you were doing the `G92 Z0`.

pandel, 2017-08-29:
@benlye That was exactly what I thought about yesterday, as I always forget about `M114` - didn't really have to use it up to now :)

I think it is questionable why `G30` behaves in such a way _now_ and how to interpret the given result. The old behaviour seems reasonable to me, as it's the "probe triggered at" Z value in relation to the current Z probe offset. I'm totally fine with that. The actual behaviour gives a value I'm unable to interpret in a meaningful way...

I didn't find anything curious while reading through the source code and so I think I will activate debug mode for levelling and see, if I can find any hint regarding this.

benlye, 2017-08-29:
I bet the output is now the 'corrected' Z location, taking into account Z offset and bed levelling correction (if enabled).  The very small numbers you get as output are simply the variance in your probe measurements.

empakoso, 2017-08-29:
Nice one! M114, also forgot about that.
I was interested in how that posting notes that there was a change in firmware regarding G30.

pandel, 2017-08-29:
@benlye Yes, might be the case, but I don't understand why, as I set `M851 Z0` before and the variable `zprobe_zoffset` should have been set correctly to 0. I suspect it could have something to do with the two probe moves (fast and slow). Perhaps I should search in this direction... but honestly, I don't have time for a deeper research at the moment :/

benlye, 2017-08-29:
Good point. Try using `M502` to make sure your firmware defaults are loaded as well.

pandel, 2017-08-30:
I did already do this every time after flashing the firmware  ;)

pandel, 2017-09-04:
Just for the record... I didn't know, that my Skynet 2.3.2 was based on Marlin 1.1.0-RCBugfix when I opened the issue. Corrected.

benlye, 2017-09-04:
Back at home and had a chance to dig into this properly.  The answer is "Yes, something changed - the `NO_WORKSPACE_OFFSETS` option was added".  

If you enable this option (found in configuration_adv.h) `G92` and `G30` will behave exactly as they used to.

My Z offset is -3.50, as found using the same process you describe (nozzle to paper, then `G92 Z0`).

With `#define NO_WORKSPACE_OFFSETS` uncommented:
```
Send: G30
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 110.00 Y: 110.00 Z: 3.50
Recv: X:135.00 Y:151.00 Z:10.00 E:0.00 Count X:13500 Y:15100 Z:4000
Recv: ok
```

With `#define NO_WORKSPACE_OFFSETS` commented:
```
Send: G30
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 110.00 Y: 110.00 Z: 0.01
Recv: X:135.00 Y:151.00 Z:10.00 E:0.00 Count X:13500 Y:15100 Z:4000
Recv: ok
```

This seems to be because G92 causes a workspace shift, which is taken into account when the probe results are returned to G30.

pandel, 2017-09-04:
Ah! Many thanks! I'll try that immediately when I'm back at my printer.

pandel, 2017-09-20:
Hey @benlye , sorry for the big delay... I tried your suggestion and it works pretty well as expected! Many thanks for that! Sadly, it doesn't solve my bilinear issues, so I went back to linear, but that's another story :)

closed 2017-09-20 by pandel