MarlinFirmware/Marlin #7578: G33 Delta Calibration Issue

plloppii, 2017-08-28:
Hello everyone,

I am having an issue on my Kossel Mini delta printer right now. When I do the G33 delta auto level command, the inductive sensor probes a point that is off the bed. 
I am using Marlin 1.1.4 and I tried adjusting the DELTA_CALIBRATION_RADIUS, however that does not affect where the sensor probes.
I have attached my config.h file.

I do have another question, I was wondering if I need to call the g33 command in the beginning of the every print? Is it necessary to call G29 for the bed mapping at all for the delta printer?

Best,
Noah

Prof62, 2017-08-29:
I had the same problem and the issue seems to be that the G33 probe area uses a constant called DELTA_PROBEABLE_RADIUS which is set lower down in configuration.h and by default is DELTA_PRINTABLE_RADIUS - 10.

This may work for the standard Allen Key probe but not if your probe is slightly farther out like my serve driven one which has a Y offset of 19mm

I fixed the problem by changing the DELTA_PROBEABLE_RADIUS to DELTA_PRINTABLE_RADIUS - 20 to suit my probe. Though with hindsight I suppose setting it to DELTA_CALIBRATION_RADIUS might be a more sensible solution.

There is also another issue in that DELTA_PROBABLE_RADIUS is only defined if MESH_BED_LEVELING, AUTO_BED_LEVELING_BILINEAR or AUTO_BED_LEVELING_UBL are defined but it is actually referenced for all levelling types:

From Configuration.h
```
#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_BILINEAR) || ENABLED(AUTO_BED_LEVELING_UBL)
  // Gradually reduce leveling correction until a set height is reached,
  // at which point movement will be level to the machine's XY plane.
  // The height can be set with M420 Z<height>
  //#define ENABLE_LEVELING_FADE_HEIGHT

  // Set the boundaries for probing (where the probe can reach).
  #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 10)

#endif

#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)

  // Set the number of grid points per dimension.
  // Works best with 5 or more points in each dimension.
  #define GRID_MAX_POINTS_X 9
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  #define LEFT_PROBE_BED_POSITION -(DELTA_PROBEABLE_RADIUS)
  #define RIGHT_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS
  #define FRONT_PROBE_BED_POSITION -(DELTA_PROBEABLE_RADIUS)
  #define BACK_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS

...etc.
```
Moving the definition for DELTA_PROBEABLE_RADIUS out of the IF block and up next to the definition for DELTA_CALIBRATION_RADIUS would solve this latter issue as well as making the parameter easier to spot.

plloppii, 2017-08-30:
Hi Prof62,

Thanks for replying, however I am still gettting the same issue. 
I have done what you said and moved the definition for DELTA_PROBEABLE_RADIUS out and next to DELTA_CALIBRATION_RADIUS, however my probe is still going off the bed. 
Do you have any other suggestions? Its extremely frustrating. 
I am using the G33 command, am I suppose to enable mesh leveling or linear leveling at all?

Here is my code,
`#define DELTA

#if ENABLED(DELTA)

  // Make delta curves from many straight lines (linear interpolation).
  // This is a trade-off between visible corners (not enough segments)
  // and processor overload (too many expensive sqrt calls).
  #define DELTA_SEGMENTS_PER_SECOND 200

  // After homing move down to a height where XY movement is unconstrained
  #define DELTA_HOME_TO_SAFE_ZONE

  // Delta calibration menu
  // uncomment to add three points calibration menu option.
  // See http://minow.blogspot.com/index.html#4918805519571907051
  //#define DELTA_CALIBRATION_MENU

  // uncomment to add G33 Delta Auto-Calibration (Enable EEPROM_SETTINGS to store results)
  #define DELTA_AUTO_CALIBRATION

  // NOTE NB all values for DELTA_* values MUST be floating point, so always have a decimal point in them

  #if ENABLED(DELTA_AUTO_CALIBRATION)
    // set the default number of probe points : n*n (1 -> 7)
    #define DELTA_CALIBRATION_DEFAULT_POINTS 3.0
  #endif

  #if ENABLED(DELTA_AUTO_CALIBRATION) || ENABLED(DELTA_CALIBRATION_MENU)
    // Set the radius for the calibration probe points - max DELTA_PRINTABLE_RADIUS*0.869 for non-eccentric probes
      #define DELTA_CALIBRATION_RADIUS 60.0 // mm
      #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 20)
  #endif
  

  // Print surface diameter/2 minus unreachable space (avoid collisions with vertical towers).
  #define DELTA_PRINTABLE_RADIUS 90.0 // mm

  // Center-to-center distance of the holes in the diagonal push rods.
  #define DELTA_DIAGONAL_ROD 215.0 // mm

  // height from z=0 to home position
  #define DELTA_HEIGHT 240.0 // get this value from auto calibrate

  #define DELTA_ENDSTOP_ADJ { 0.0, 0.0, 0.0 } // get these from auto calibrate

  // Horizontal distance bridged by diagonal push rods when effector is centered.
  #define DELTA_RADIUS 107.0 //mm  Get this value from auto calibrate
  
  // Trim adjustments for individual towers
  // tower angle corrections for X and Y tower / rotate XYZ so Z tower angle = 0
  // measured in degrees anticlockwise looking from above the printer
  #define DELTA_TOWER_ANGLE_TRIM { 0.0, 0.0, 0.0 } // get these values from auto calibrate

  // delta radius and diaginal rod adjustments measured in mm
  //#define DELTA_RADIUS_TRIM_TOWER { 0.0, 0.0, 0.0 }
  //#define DELTA_DIAGONAL_ROD_TRIM_TOWER { 0.0, 0.0, 0.0 }

...

#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_BILINEAR) || ENABLED(AUTO_BED_LEVELING_UBL)
  // Gradually reduce leveling correction until a set height is reached,
  // at which point movement will be level to the machine's XY plane.
  // The height can be set with M420 Z<height>
  //#define ENABLE_LEVELING_FADE_HEIGHT

  // Set the boundaries for probing (where the probe can reach).
  //#define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 20)

#endif

#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)

  // Set the number of grid points per dimension.
  // Works best with 5 or more points in each dimension.
  #define GRID_MAX_POINTS_X 9
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  //#define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 23)

  #define LEFT_PROBE_BED_POSITION -(DELTA_PROBEABLE_RADIUS)
  #define RIGHT_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS
  #define FRONT_PROBE_BED_POSITION -(DELTA_PROBEABLE_RADIUS)
  #define BACK_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS

  // The Z probe minimum outer margin (to validate G29 parameters).
  #define MIN_PROBE_EDGE 10

`

punamenon, 2017-09-11:
There is a value called DELTA_CALIBRATION_RADIUS on line 473 of configuration.h 

If you change that value it will adjust the radius of your G33 calibration points and nothing else.

mshicom, 2017-12-20:
After you changed the DELTA_CALIBRATION_RADIUS parameter, you need to refresh the EEPROM by using the gcode command M502 and M500, otherwise the old settings will remain. Actually you can change this parameter online by using M665 B70 (e.g. with DELTA_CALIBRATION_RADIUS being 70).

mtraven, 2018-04-11:
I am currently working on calibrating my scratch built delta printer when I noticed g33 does not use the xy offset used elsewhere..That led me to this thread...I guess I am wondering why?  Is there a good reason for this, or is this more of a bug/lack of feature.  It seems to me that this is likely to cause errors in calibration, especially on beds that have been machined (errors would likely be concentric).  Is anyone working on correcting this?  I am thinking about having a go at it, but my skillset it much more suited for the construction of the printer, not the coding....

thinkyhead, 2018-04-12:
@mtraven — Which XY offset are you referring to? The one set by `G92`?

closed 2018-04-12 by thinkyhead

thinkyhead, 2019-02-19:
An offset probe simply isn't suitable for all of the calibrations available in `G33`, though it may work for some of them. As explained, the reason is that the current positions of the arms is used. An offset probe makes that impossible.

yashunandansureka, 2019-03-04:
I am having a similar issue, i downloaded a copy of the marlin 1.1.9, and configured it to my delta, now when i go for the G33, the probe goes down and hits the bed, motors stall and then i get error, probing failed.

Bashtarle, 2019-03-08:
@yashunandansureka ran into a similar issue today when I was swapping out my drivers because they for whatever reason are reversed from my previous drivers. So Instead of going up to preform the correct homing procedure it was going down and slamming into the bed. 

So it sounds like you might be having a similar issue where your old configuration inverted the motor directions in software and those settings didn't get carried over. At least I know going in and inverting the directions solved my issues with the new driver boards. 

Unless it's homing fine in which case I have no idea.