MarlinFirmware/Marlin #7581: [FR] Support for M290 - baby stepping

dosono, 2017-08-29:
Hi,
As far as I know the only support for great "baby stepping" feature  is from the LCD controller.
Since I am using a touch screen (MKS) I am trying to find a way to get this featurein the form of gcode  (you can configure a gcode related button).

I can see that there is a gcode for that: 
http://reprap.org/wiki/G-code#M290:_Baby_stepping
but it says that it is not supported in Marlin.  

So, is it truly not supported in Marlin or is it truly a new feature request ?

pandel, 2017-08-29:
Doesn't seem it is supported at the moment: http://marlinfw.org/meta/gcode/

tcm0116, 2017-09-07:
One option would be to enable `BABYSTEPPING` and `BABYSTEP_ZPROBE_OFFSET` in Configuration_adv.h, and then you could use `M851` to adjust the Z Offset which will babystep the Z axis. I believe there is an iissue where the Z axis does not babystep when using bilinear bed leveling, so you'll want to consider that.

dosono, 2017-09-09:
Interesting. So you say I can use this during prints.... I will give it a try.

closed 2017-10-19 by thinkyhead