MarlinFirmware/Marlin #7584: Marlin 1.1.4

RobertsClif, 2017-08-29:
I have a unique problem  It involves "Endstop Trigger reporting"

I get an error Using a Proximity sensor TL_M3W 'PNP' as a Z-Axis endstop.
using these lines of code.

#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_CLEARANCE_DEPLOY_PROBE   10  // Z Clearance for Deploy/Stow

#define X_PROBE_OFFSET_FROM_EXTRUDER 10     // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 25     // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.5   // Z offset: -below +above  [the nozzle]

This allows me to home then raise Z-Axis 10mm for a clearance of 10.5mm 
When I Auto home everything works up to home, the "Z Clearance for Deploy/Stow" parts finishes I get the error " Endstops Z ".  I Believe that because the Z-Axis homed, then rose to the Clearance is why I am getting the error "endstop still in motion".
 
Now when I use this line

#define Z_MIN_PROBE_ENDSTOP

I get no errors, though I can not get the Z-Axis to Raise to my wanted clearance of 10.5mm.

This works completely in Marlin v1.1RC8 renamed to HICi3 v1.1RC8

RobertsClif, 2017-09-08:
Today I re-flashed both 3D Printer with Marlin 1.1.5, 
Now one works one on still displays Display " Endstops Z ", Before you say it I swapped the Controller that did not work with a spare controller is does not work. same error.

RobertsClif, 2017-09-09:
Here is the latest testing results.

Raise bed to 100mm and Auto home, after the X and Y Axis home we single tap the Z-Proximity sensor with a metal putty knife simulating the bed.
It will detect what it believes is bed rise 10mm then drop down again to what it believes is Z-Axis zero, once more will rise to Z_CLEARANCE_DEPLOY_PROBE 10 this time it has no display to " Endstops Z " error.

Have done the dozens of times with no " Endstops Z " error, as soon a it detects the bed the second time I get the Error Again.

What causes the Bed to Drop the Second time.

Roxy-3D, 2017-09-09:
You can send a M119 to get the current status of the end stops.   That is very helpful because you can manually trigger one and send the command and see what the firmware thinks is the end stop status.

RobertsClif, 2017-09-10:
I can do an entire 25 Point Bed leveling, after Receiving the " Endstops Z " on the first test point, complete the Entire Print. I  believe the TL_M3W 'NPN' NC  Proximity sensor works.

RobertsClif, 2017-09-15:
I am still on Marlin 1.1.X will be trying Marlin till I find where it starts erring. I know Marlin  1.1.1 and above errors.

    Compiled Marlin 1.1.1 displays " Endstops Z " error however no CRC memory error
    Complied Marlin 1.1.2 displays " Endstops Z " error and A CRC memory error -8370 :
//Crazy error actually Marlin 1.1.2 is using 64 bytes less memory than Marlin 1.1.1
forgot to mention both display in pronterface "echo:endstop hit: Z0.50"

Does anyone know where I can download just the updated code changes for these versions from Marlin 1.1.0 to Marlin 1.1.1 to Marlin 1.1.2

Thank You

RobertsClif, 2017-09-18:
Don't believe this is an error now, looks like it is just a checksum that is stored with the data for testing. 

------------------------------------------------------
#define DUMMY_PID_VALUE 3000.0f
#define EEPROM_START() int eeprom_index = EEPROM_OFFSET
#define EEPROM_SKIP(VAR) eeprom_index += sizeof(VAR)
#define EEPROM_WRITE(VAR) write_data(eeprom_index, (uint8_t*)&VAR, sizeof(VAR), &working_crc)
#define EEPROM_READ(VAR) read_data(eeprom_index, (uint8_t*)&VAR, sizeof(VAR), &working_crc)
#define EEPROM_ASSERT(TST,ERR) if (!(TST)) do{ SERIAL_ERROR_START(); SERIAL_ERRORLNPGM(ERR); eeprom_read_error = true; }while(0)
------------------------------------------------------
const uint16_t final_crc = working_crc;
------------------------------------------------------
EEPROM_READ(stored_crc);

RobertsClif, 2017-09-18:
Don't believe this is an error now, looks like it is just a checksum that is stored with the data for testing. 

------------------------------------------------------
define DUMMY_PID_VALUE 3000.0f
define EEPROM_START() int eeprom_index = EEPROM_OFFSET
define EEPROM_SKIP(VAR) eeprom_index += sizeof(VAR)
define EEPROM_WRITE(VAR) write_data(eeprom_index, (uint8_t*)&VAR, sizeof(VAR), &working_crc)
define EEPROM_READ(VAR) read_data(eeprom_index, (uint8_t*)&VAR, sizeof(VAR), &working_crc)
define EEPROM_ASSERT(TST,ERR) if (!(TST)) do{ SERIAL_ERROR_START(); SERIAL_ERRORLNPGM(ERR); eeprom_read_error = true; }while(0)
------------------------------------------------------
const uint16_t final_crc = working_crc;
------------------------------------------------------
EEPROM_READ(stored_crc);

closed 2017-11-14 by thinkyhead