MarlinFirmware/Marlin #7588: Marlin 1.1.5: Bed Size, Home and UBL

Sineos, 2017-08-31:
I have some trouble understanding the Bed Size concept in relation to Home and UBL:

- My endstops do not correspond to edge of the bed
- My bed actually starts at X=25, Y=6
- Same is true for the back end of the bed with respect to Y

Problem:
- When I start UBL it tries to probe outside of the bed
- My prints are off center

What is the correct way to harmonize bed edges, bed center and UBL bounderies:
- Can it be offsetted with M206?
- Is MANUAL_X_HOME_POS / MANUAL_Y_HOME_POS the right way?

Any hints would be appreciated.

Roxy-3D, 2017-08-31:
BED_SIZE was added because in general it makes it easier to configure the printer.   But if that is not true (as in your case), you can still use the X_MIN_POS way to configure things.

Alternatively...   You can still use BED_SIZE, but add an INSET to make the machine avoid the coordinate space that is 'off the bed'.

ellisgl, 2017-09-22:
Mine - I have issue with the Y, but Y_MIN_POS doesn't change a thing.  I home Y and it stays at 0. I've tried messing with it in the Firmware and in the eeprom, but same thing...

Provide an example?

fiveangle, 2017-09-22:
> My bed actually starts at X=25, Y=6

No, your bed must always start at 0,0 so think of it as your movement constraints extend to -25, -6.  Make more sense in that context ?

-=dave

ellisgl, 2017-09-23:
Y_MIN_POS -## doesn't change it. It's Y and is still 0. 
MANUAL_Y_HOME_POS stays against the sensor, but software will state it's what ever you put in there.

ellisgl, 2017-09-23:
Z_SAFE_HOMING it is.

closed 2017-11-27 by thinkyhead