MarlinFirmware/Marlin #7590: [FR] / Help - Repeat last print via gcode macro

gordo3di, 2017-08-31:
I LOVE the new custom user gcode macro section in the new 1.1.4 snapshot. One of the features we tried implementing before is repeat last print and I think this can be done with a user gcode macro now.

The problem i'm running into is trying to somehow save the path to the last gcode file printed so i can then use M23 and M24 to just simply repeat that print. For production printers this feature would save so much time.

I've already tried to save the path to a new variable in cardreader.cpp both at the end and when the print job is initialized but no success. If you have any helpful hints on how to accomplish this it would be so useful. I feel like its an easy thing to accomplish but I've already spent about 8 hours messing with the code to no avail.. Thanks for any help and again we love Marlin!

Heres the fails i managed to rack up in 

  #define USER_DESC_4 "Repeat Last Print"
  #define USER_GCODE_4 "M117 " STRINGIFY(repeatlastprint) "\nM0\n;\n;\n;\n;\n"
  //#define USER_GCODE_4 "M117 " STRINGIFY(lastSDfilename) "\nM0\n;\n;\n;\n;\n"
  //#define USER_GCODE_4 "M32 P !fast.gco#" // works directly
  //#define USER_GCODE_4 "M23 fastprint.gco\nM24" // didn't work

I think i can save the path somewhere in here from cardreader.cpp:

void CardReader::openAndPrintFile(const char *name) {
char cmd[4 + strlen(name) + 1]; // Room for "M23 ", filename, and null
sprintf_P(cmd, PSTR("M23 %s"), name);
for (char *c = &cmd[4]; *c; c++) *c = tolower(*c);
// #define repeatlastprint cmd // my addition
enqueue_and_echo_command(cmd);
enqueue_and_echo_commands_P(PSTR("M24"));
}

Thanks for any help

Roxy-3D, 2017-08-31:
Gordon...   Sit tight...   One way or the other...   We'll get this feature into the main code base for you.

gordo3di, 2017-08-31:
Haha I love you Roxy!

Roxy-3D, 2017-09-02:
@gordo3di     Gordon...   I think I have it close.    But if not, please say what needs to be tweaked and I'll try to get it perfect for you.      The branch is at https://github.com/Roxy-3D/SD-Reselect-and-Print    Thinky is trying to do the File and Folder reorganization so I don't want to merge this into the main branch until that is done.  It will just cause chaos for him.

But you can go ahead and see if these changes do what you want.   This was tested on my gMax 1.5+ but I have UBL turned on.   I did not commit the UBL enablement to this branch, but you should be aware it wasn't tested in your preferred environment. 

All you need to do is turn on this option for this feature to be active.   (And it is already turned on in the main Configuration_adv.h file of this branch:
```cpp
  // This option allows you to easily reprint the same SD Card file as
  // was last printed.  At the end of a print, the LCD Menu will jump
  // straight to file previously selected.  A single click of the encoder
  // wheel will restart the print.  Another file or LCD option can be 
  // selected by using the encoder wheel to navigate through the menu structure.
  #define SD_REPRINT_LAST_SELECTED_FILE
````
The LCD Panel will not time out if the screen is trying to select an SD File.   That is so when the printer stops and puts up the SD File Menu...  It can sit for 10 minutes while you remove the current print and prep the bed for the next one.   Other than that...  The LCD Menu screens will still time out if the panel sits idle too long.

PS.  I understand how you spent 8 hours working on this and did not get it going.   It is amazingly simple (and only a few lines of code) if you understand everything.   But I didn't understand everything about the LCD Macros when I started and things get complicated fast.

gordo3di, 2017-09-02:
Awesome. Thanks Roxy and Scott. I will dl it today and check it out. I figured it wouldn't be that hard for a proper programmer. I can program but i tend to swim in circles sometimes. Is this branch based off 1.1.5+ or is it the 1.1.4 branch?

Roxy-3D, 2017-09-02:
It is forked off of the bugfix-v1.1.x branch two or three days ago.   It should be very current.   And once the file and folder reorganization is done...  I'll move the changes to the main branch.

gordo3di, 2017-09-02:
Awesome. The new releases seem to work quite well. Much more responsive LCD
and bed leveling. I had an issues compiling with the filament change
feature but I haven't tried the latest yet.

Roxy-3D, 2017-09-02:
A couple more small changes got added to the branch.   Every once in a while the status screen did not clear and get repainted with the SD File Select screen.   The SD File Select screen was active, and a click would still restart the print.   But... in the latest code that problem is gone.

gordo3di, 2017-09-03:
Nice. Working late so ill be here if i can help at all.

Tannoo, 2017-09-03:
>```
  // This option allows you to easily reprint the same SD Card file as
  // was last printed.  At the end of a print, the LCD Menu will jump
  // straight to file previously selected.  A single click of the encoder
  // wheel will restart the print.  Another file or LCD option can be 
  // selected by using the encoder wheel to navigate through the menu structure.
  #define SD_REPRINT_LAST_SELECTED_FILE
```

I can't seem to find it in that fork. Is there another branch in that fork that has this?

Tannoo, 2017-09-03:
Was this reprint option not committed?

Roxy-3D, 2017-09-03:
It's not in the main Marlin code base yet.  I don't want to add any chaos and confusion the file and folder layout.   Once that happens...  I'll move this handful of lines of code change over to the main branch.   But for now...  It only exists as a one off branch.

Tannoo, 2017-09-03:
I realize that... I can't find it on your GitHub.

Roxy-3D, 2017-09-03:
There is something weird going on.  I looked at the modification dates and ultralcd.cpp has not been modified for 17 days.

I'll figure it out in the morning.  I tried to make a .ZIP file and GitHub complained it was too big.   I was going to just give you a .ZIP file so you could see what I did.   (Its a Saturday night and I was playing darts....  I am not competent to do complicated computer things right now.)

Roxy-3D, 2017-09-03:
> Was this reprint option not committed?

It is there now.

Roxy-3D, 2017-09-03:
@Tannoo    I think I need help with this.    Usually the SD File Select screen appears at the end of print.  But not always.    When it does not appear (and the status screen is still showing) just clicking the encoder wheel on notch makes it appear and everything is right.

For some reason, the LCD Update routine is not redrawing the screen even though it is being told to do that.   For quick and easy testing, I added some  short gcode files that just move the nozzle in a square path.   And at the end of them...  They should pop up the SD File Select screen (if launched from the LCD Panel)

[Gcode.zip](https://github.com/MarlinFirmware/Marlin/files/1273043/Gcode.zip)

Tannoo, 2017-09-03:
>Usually the SD File Select screen appears at the end of print. But not always. When it does not appear (and the status screen is still showing) just moving the encoder wheel one notch makes it appear and everything is right.

I was having the same exact issue with the lighting menu I did. No lcd update commands were doing what was implied.

Roxy-3D, 2017-09-03:
The LCD Code needs to be re-written.  But there must be a way to force the update.    I'm wondering if this only happens the Graphical LCD's.   I'm wondering if this is a side effect of the update being split into multiple parts and it only doing some of the update each cycle.

Tannoo, 2017-09-03:
I am trying it on the Re-ARM with the 20x4 display. After the files are done, the display is blank. No amount of turning the encoder makes it come back.

Roxy-3D, 2017-09-03:
Obviously...  There is something I don't understand about the LCD code.   But what I do fully understand is the LCD code needs to be re-written!

Tannoo, 2017-09-03:
Thinky has stated that it needs to be re-written, also.

Tannoo, 2017-09-03:
I know that `lcd_goto_screen` already updates the lcd when it changes screens.

Roxy-3D, 2017-09-03:
Yeah...   In a perfect world...   But once in a while... it doesn't do its thing...

Tannoo, 2017-09-03:
I'm trying to find the updates that are stepping on each other. Because for me, the sd menu screen is blank once the file has been printed.

Roxy-3D, 2017-09-03:
And something else for you to keep in your head while you are looking at the LCD code:   I added some brute force code trying to work around things.   It didn't really help:
```cpp
void CardReader::printingHasFinished() {
  stepper.synchronize();
  file.close();
  if (file_subcall_ctr > 0) { // Heading up to a parent file that called current as a procedure.
    file_subcall_ctr--;
    openFile(proc_filenames[file_subcall_ctr], true, true);
    setIndex(filespos[file_subcall_ctr]);
    startFileprint();
  }
  else {
    sdprinting = false;
    if (SD_FINISHED_STEPPERRELEASE)
      enqueue_and_echo_commands_P(PSTR(SD_FINISHED_RELEASECOMMAND));
    print_job_timer.stop();
    if (print_job_timer.duration() > 60)
      enqueue_and_echo_commands_P(PSTR("M31"));
    #if ENABLED(SDCARD_SORT_ALPHA)
      presort();
    #endif
 }
 #if ENABLED(SD_REPRINT_LAST_SELECTED_FILE)
// lcdDrawUpdate = LCDVIEW_REDRAW_NOW;
   drawing_screen = true;
   lcdDrawUpdate = LCDVIEW_CLEAR_CALL_REDRAW;
   lcd_goto_screen(lcd_sdcard_menu, saved_encoderPosition);
   lcd_goto_screen(lcd_sdcard_menu, saved_encoderPosition);
   lcd_goto_screen(lcd_sdcard_menu, saved_encoderPosition);
   screen_changed = true;
//    #if ENABLED(DOGLCD)
//      drawing_screen = false;
//    #endif
 #endif
}
```

Tannoo, 2017-09-03:
Ok.

Tannoo, 2017-09-04:
I have it working, but the LCD is still blank. 2 clicks make it print the same file again.

Roxy-3D, 2017-09-04:
What branch is it in?

Roxy-3D, 2017-09-04:
Is it a Re-ARM problem that the screen is blank?   Or is it a code problem?

Tannoo, 2017-09-04:
I dunno yet... I think it just may that nasty spi thing hanging around the Re-ARM.

Tannoo, 2017-09-04:
>What branch is it in?

This branch is based from `bugfix-2.0.x`.
https://github.com/Tannoo/Marlin/tree/My_Re-ARM

Tannoo, 2017-09-04:
I haven't touched my printer or it's code in a while. I should put it on that one and see what happens.

Roxy-3D, 2017-09-04:
So far, I've only been working on this on a Graphical LCD.  I should try it on a 20x4 and see if the same problems happen.  That might help us isolate the root cause.

Tannoo, 2017-09-04:
Ok... see my updated post..  /\

Tannoo, 2017-09-04:
If I click to get it back to the info screen just before it stops "printing", the status message changes to `Click to resume...`.

Tannoo, 2017-09-04:
I click to the resume, and the file prints all over again.

Roxy-3D, 2017-09-04:
Oh!   Then that is a bug in the new code.   I hooked into the void CardReader::printingHasFinished() function to re-activate the SD File Select menu.   Apparently, that gets called when a user pauses the print also.   So....  The code needs to know whether it is a pause or a completed print!   

Because right now I have void CardReader::printingHasFinished() doing the    lcd_goto_screen(lcd_sdcard_menu, saved_encoderPosition); to re-activate the file select menu screen.

Tannoo, 2017-09-04:
Oh, I know.... I'm just flying blind.. lol

Tannoo, 2017-09-04:
I had to make some small changes to get AVR to compile. Going to test it soon.

Tannoo, 2017-09-04:
Okay, on the Mega (AVR), it works as intended, but, it takes 2 clicks to print again.
It displays the sd file menu and the printed file is selected.
The first click doesn't do anything.
The second click prints the file again.

I tried this several times on the same file, and it's the same thing.

Tannoo, 2017-09-04:
FYI, this is the same bugfix-2.0.x branch that I have on the Re-ARM. I just changed the MB and the pertinent stuff for the printer.

Tannoo, 2017-09-04:
The first "dead" click, only sounds the click beep. But nothing else changes.
If I change the file selection before the first click, that first click does NOT change `saved_encoderPosition` value.
The second click will change it.

Tannoo, 2017-09-04:
I don't have any display issues on the full graphic display.

Tannoo, 2017-09-04:
I created an AVR branch that is currently on my printer:
https://github.com/Tannoo/Marlin/tree/My_AVR

Tannoo, 2017-09-04:
Also based on bugfix-2.0.x

Roxy-3D, 2017-09-04:
I think you already realized this...  But I just got there...    The 20x4 and the Graphical LCD have different code to do the static void lcd_implementation_status_screen() function.   When I said the LCD stayed on the status screen...  That is the information that is on the display.  The LCD Panel actually has the SD Card Select menu active.  Any movement of the encoder wheel refreshes the screen and it will list files.

In your case on the 20x4 LCD...  The screen is clear.  But almost for sure the SD Card Select menu is also active.    The fact we are getting different information on the LCD display might help us figure out how the failure is happening.

Tannoo, 2017-09-04:
1) I don't have any issues with the LCD (reprap full graphic) on the AVR.
The issue I DO have on the AVR, is an extra click when this reprint feature is enabled.

2) On the Re-ARM... all kinds of display 20x4 (reprap discount) issues while accessing the lcd's SD port.
This is not related to your reprint code.

If the extra click is found and eliminated, I believe the code will be good for main stream.

Roxy-3D, 2017-09-06:
We had an extra click problem back when you first started working on the LCD code.   We must have resolved that.   But now you are seeing it again?

Tannoo, 2017-09-06:
Yeah, I think that was when getting the click to edit mesh points from the ubl lcd map to work.

Tannoo, 2017-09-07:
I moved the `cardreader.cpp` part of the code to `marlin_main.cpp`
```
        if (card_eof) {
          SERIAL_PROTOCOLLNPGM(MSG_FILE_PRINTED);
          card.printingHasFinished();
          #if ENABLED(SD_REPRINT_LAST_SELECTED_FILE)
            lcd_goto_screen(lcd_sdcard_menu, saved_encoderPosition);
            #if ENABLED(MAX7219_DEBUG_SD)
              Max7219_encoder_position();
            #endif
          #endif
          #if ENABLED(DONE_PRINTING_RGB)
            LCD_MESSAGEPGM(MSG_INFO_COMPLETED_PRINTS);
            #if HAS_RESUME_CONTINUE
              set_led_color(0, 255, 0, 0); // Green
              //enqueue_and_echo_commands_P(PSTR("M0")); // end of the queue!  <<-- CULPRIT!
            #else
              safe_delay(1000);
              set_led_color(0, 0, 0, 0);   // OFF
            #endif
          #endif
          card.checkautostart(true);
        }
```

Tannoo, 2017-09-07:
It now works as intended. That M0 command is NOT the end of the queue. There are still 2 more moves needing to be made.

gordo3di, 2017-09-07:
This is great.  Gonna start playing with it tomorrow

Roxy-3D, 2017-09-07:
> It now works as intended. That M0 command is NOT the end of the queue. There are still 2 more moves needing to be made.

Wow!   I was confused why the Max7219 Stepper Queue lights always blinked 2 extra commands at the end.   Now I know.   If you have your Max7219 plugged in and turned on...  You will see what I mean.

Tannoo, 2017-09-08:
> If you have your Max7219 plugged in and turned on... You will see what I mean.

I do and have.

Roxy-3D, 2017-10-26:
I have merged a fully tested (and debugged) version.  It is merged into
both the bugfix-v1.1.x branch and the bugfix-v2.0.0 branch.

You just enable the #define SD_REPRINT_LAST_SELECTED_FILE option in
Configuration_adv.h and you should be good to go.

There is an unresolved issue that it uncovered.   When scrolling up or down
in a list of SD Files, it is possible for the 'Selection Bars' to not get
cleared  and to be repainted on the new file.   This isn't a problem with
the Auto Re-Print.   It is a problem with the way the LCD Update code is
trying to refresh only 1/2 the screen at a time so the Graphical LCD
Displays don't consume the entire CPU.    You can ignore that issue for
now.   We will get that figured out.    (And it only happens if you spin
the encoder wheel fast.  There is no encoder wheel movement on the reprint)

closed 2018-02-09 by thinkyhead