MarlinFirmware/Marlin #7594: [FR] Shutoff all heaters after X minutes or hours of inactivity

marcio-ao, 2017-08-31:
It would be nice to have the ability to shut off all heaters after X minutes of inactivity. This value should be user customizable either through the LCD or an Mcode (options may be 30 minutes, 1 hour, 3 hrs, etc). This would be useful because certain slicers set the bed temperature to a part removal temperature, but if the user doesn't come to remove the part, it would be nice to eventually power down the heaters both for safety and for energy conservation.

This would be similar to DEFAULT_STEPPER_DEACTIVE_TIME, but for the hot end and bed heaters.

gordo3di, 2017-08-31:
Wee set it in the end gcode. So the user determines it.

cdedwards, 2017-08-31:
Thought about adding this myself. great idea, as I have discovered my hotend and bed still hot over night.

marcio-ao, 2017-08-31:
@gordo3di: Yes, the part removal temperature is set in the end GCODE, but the printer will maintain that temperature indefinitely. It would be nice if the there was a timeout value that would cause all the heaters to turn off after a certain amount of time of inactivity.

marcio-ao, 2018-08-20:
I am closing this FR since I was told that allowing the bed to cool completely could damage the print bed surface. So I am no longer looking to implement this timeout.

closed 2018-08-20 by marcio-ao