MarlinFirmware/Marlin #7595: [FR] Scroll large file names into view

marcio-ao, 2017-08-31:
When there are files with long names in the SD card, the name is currently truncated and it is difficult to tell which file you are selecting. It would be nice if the currently highlighted filename would scroll into view so you could read the rest of the name.

I would be willing to attempt to implement this feature myself, but I would need to some pointers. I don't quite understand how the menus are drawn and in particular how I would go about implementing an animation loop. Perhaps if someone could point me in the right direction I could give it a shot.

Thank you,

-- Marcio

gordo3di, 2017-08-31:
They have it for LCD notifications now as of 1.1.4. It would be amazing for long filenames. I tried it years ago but never made it.

marcio-ao, 2017-09-05:
@gordo3di : I just published a working implementation of this for this for review :)

GMagician, 2017-11-18:
This can be closed, I think

closed 2018-02-09 by thinkyhead