MarlinFirmware/Marlin #7597: Understanding  FIRMWARE RETRACTATION G10 / G11 / G10 S1

studiodyne, 2017-09-01:
To be googlized for many users that need to be informed about this functionnality

**What is it  ?**

> It's a functionnality , that permit to ask your printer to make the retraction and restart. The advantages is that you can change parameters during printing , and see with you eyes what effect the parameters have , it's very usefull to find the exact settings to restart the more clean as possible. Your slicer just say G10 , and the machine will execute the retractation with its own settings or say G11 and the machine recover , and G10 S1 for long retract 'on tool change '.

**Stop your illusions !! Why?**

> Firmware retraction , is just a tool , an utility and will never be the future of printing , for many reasons. This utility is just recommended to test a printer , to test some parameters to know more about your nozzle reactivity. But the big problem of firmware retract is the primitive and robotic movement , that will never help to make a beautifull print. You know that a retraction oozes and the only way to solve this problem , is to wipe the extruder and in the same time to control extrusion , this way need 2 or 3 movement at the same time , z go down ' with zlift activated' , x and y move to wipe , and extruder that runs according to the other movements. This smooth smart wipe is only possible to be made by the slicer , because it's the generator of the toolpath and it's the only system that can calculate a seamless printed object , it has the object and it calculates all nozzle movements the best as possible.
> It's the reason why , firmware retract have no interest for the slicers , too primitive to help to make smooth toolpath.  The algorithm of a smooth firmware retractation will never have the intelligence to find the toolpath to wipe the object , the machine doesn't know the object and doesn't have it in its memory 

:-)
Ultimaker use only G10 and G10 S1 , but no G11 , G11 is brutality for a print , and an interpolate movement is used for restoring extrusion

How to use it 👍

Simplify3d ; You have to script for replacing the retractation instructions , before , you will make some special setting to simplify the identification by the script

Cura : coming soon , because i'm in war with cura , not made for it and need to be scripted but not my favorite slicer , i'm working on it . The option ' volumetric marlin' make only G10 and no G11 , i'm sorry , i will have more information soon

Slicer : Coming soon , i'm on it now

All help is welcome to make an instructive topic about Fw

Thks

lordofhyphens, 2017-09-02:
I don't think this is really applicable as an issue against Marlin. Also, Slic3r 1.3.0-dev has pre- and post-toolchange custom gcode.

studiodyne, 2017-09-02:
If i write to you , it’s because of the problem stated in my marlin ‘ firmware topic ‘

If i use your ‘ script tab’ , and place a G10 S1 on the tab , on the produced Gcode i will have

G10 ; retract     ‘ the normal retract ‘
G10 S1              ‘ my added gcode ‘

It’s not serious , to confuse the machine with 2 instructions , and tweaking marlin just for a bad gcode production is not serious too

It’s why , you have to make a serious Firmware retractation gcode generation  by replace any G10 before a tool change by a G10 S1
Or find a solution as you want , and the more easy as possible for you

For the moment , SLIC3R not ready and not scriptable for Multi extruder and Firmware retractation

Thks to understand , if i write you , it’s because i have big experience and no solutions to give for your users

Best regards

studiodyne, 2017-09-02:
I have a good news  @lordofhyphens  develloper of SLIC3R have accepted to work on this and have made a PR , to make a full comptability for MARLIN !!! Very good news ! Go on the link below you will have the page with my suggestion , and the PR he have made for MARLIN.
@thinkyhead  On my demand , @lordofhyphens  dev of slic3r have made a PR to have full compatibility with Marlin Firmware Retraction , now simplify3d is not the only software that make it !!!!

lordofhyphens, 2017-09-02:
It's currently a pull request https://github.com/alexrj/Slic3r/pull/4108. 

If this behavior is in a release version of Marlin I will be able to merge it. 
This might turn into a Marlin-specific gcode flavor, depending on what else breaks because of this syntax change (if other RepRap firmwares ignore the S1 then great).

lordofhyphens, 2018-01-06:
@thinkyhead did the stuff for this get into Marlin proper (so it's safe to make it part of Slic3r's gcode generation for it)?

thinkyhead, 2018-01-08:
@lordofhyphens We have had `G10`/`G11` for a while, with various patches along the way. Try 1.1.8, bugfix-1.1.x, or wait for the imminent release of 1.1.9.

thinkyhead, 2018-01-08:
> did the stuff for this get into Marlin proper

@lordofhyphens We've had `G10`/`G10 S1`/`G11` for a while, with various patches along the way. Try 1.1.8, bugfix-1.1.x, or wait for the imminent release of 1.1.9 and see if it works as advertised.

> need a little correction of MARLIN FWretract to be perfect

Ah. Well, apparently @studiodyne has some idea that there are things to change, but (s)he needs help with Github to make it happen. I have too much on my plate to write and test the code myself today. See #7607 to see what (s)he thinks is needed.

studiodyne, 2018-01-08:
History
-Marlin FwRetract was not really used except by me
-Marlin FwRetract was young and not really Tested and Ensured
-I learn C++
-I make a proposal and a PR that was accepted by thinkyhead
-In parallel i had a big problem with slicer that i have asked to solve and was solved

Now all is solved and functionnal

Details
Marlin Fw was corruptable , and i have made a ' common sense ' engine , for example , when G10S1'long retract' all g10 are not allowed ' common sense ' and more like this
But the second problem was ' SLIC3R ' 
Why ?
Because Fwretract is young , and not implemented in all slicers for the moment , excepted slicer and ' simplify if batched '
The problem with slicer is the ' NOT SEPARATE' retraction commands for example 
G1 X  Y  E22 ' extrusion'
G1 X Y   E19  ' Retraction of 3mm '  , in this cas , impossible to batch , because impossible to detect it in the gcode file 
In simplify3d  retractation is SEPARATE , 
G1 X Y E10
G92 E0
G1 E-3
G92 E0  with this gcode it's perfect to batch
Then , when i have tested slicer , and have checked the box ' fw retract ' , i had a suprise , that the long retract not implemented
I have asked the boss of slicer , to make a special FLAVOR JUST FOR MARLIN FLAVOR SELECTION
And he had made a good job , but , he told me that it's impossible to delete a ' LITTLE G10 ' that appears sometimes after G10S1

but Marlin is protected with ' no common sense' commands

Breaks ? No ! because , slic3r have G10S1 JUST FOR US ! yeah
If you use marlin , the recent release have an intelligent Fw Retractration 
Tested ? Every day by me

Thks
AMEN !

studiodyne, 2018-01-08:
Marlin is simply , the ONLY ONE firmware that have a Natural/smart FwRetract engine , and is now a ' standard reference ' for all slicers devs 😄

lordofhyphens, 2018-01-08:
@thinkyhead I don't use Marlin myself, I am asking as a Slic3r dev. 

@studiodyne has a pull request open in Slic3r to change the gcode generation for fw retract to reflect this syntax.

I will not merge changes in function in Slic3r for Marlin if those aren't in the dev tree for Marlin.

studiodyne, 2018-01-08:
@lordofhyphens Are you serious ?  This is the third stable release of marlin that have the ensured FwRetract inside...If you follow all the links here , you will see the days and the nights we have spent for just 'one right G10S1' at the right place , but , it was impossible to delete the original G10 after G10S1 , your dev said , too much work to delete this unwanted G10 after G10S1 
But it's not a problem , marlin cancel a G10 command after G10S1
If you want to make something amazing , delete this G10 that appears after G10S1 
But if you want to burn the days and night we have spent in test ' look in the links you can follow each test after each releases and all will works after a lot of slicers releases
But i'm not the boss , i'm not dev , i'm not engineer , i have no high status here and my english is so bad that i have difficulties to understand when someone tell me something technical, i'm here and i contribute with only one tool ' the good sense ' it's my favorite direction
You're afraid to break something , i understand 
I'm not boss , so it's my last message , i have given you all  informations about this 

Thks
Steeve
I just contribute

lordofhyphens, 2018-01-08:
All I wanted to be sure about is that the current "long retraction" syntax is indeed in Marlin proper. I don't follow Marlin development much at all.

Because if it wasn't, I am the one who is going to  have to deal with people saying I "broke" FW retraction for them on Marlin.

lordofhyphens, 2018-01-08:
@thinkyhead Thanks for the response. 

I looked over the merged PRs and  by @studiodyne and frankly couldn't tell what is going on. 

Looking over your own docs it looks like the syntax is there. I'll merge the PR I made to support this syntax over at https://github.com/alexrj/Slic3r/pull/4108

studiodyne, 2018-01-09:
@lordofhyphens 
You can merge , but you can improve by deleting this undeletable G10 after G10S1

-Yes Marlin Have G10 S1 syntax , and it's not a new syntax , it was implemented a few monthes or years ago
- Yes Marlin support G10 S1 long retract , and have LCD menus for feedrate and length separately of short retract 

On the marlin side , all is ok , you can merge , but , if 'possible ' to have a beautifull flavor ' by deleting the G10 that appears automaticly after G10 S1 it's a very good progress for slic3r
For the moment , slic3r 'if merged ' is the only one software that have a total firmware retractation ' only for marlin :-) ' . If you have time to work on it :-) and to complete the PR to avoid another one in the future  , it's perfect

studiodyne, 2018-01-09:
@lordofhyphens , I want to suggest an idea
You have implemented the ' variable layer Height' ok , but , if i was the author , i should make another way , and I want to suggest it to you

Outlines height and infill height ' SEPARATE'
The software build the outlines at high quality '0.1 ' and after 4 slices , you print the infill with ' 0.4 '
We have now a very good looking for the same time of printing in low quality :-)
You just have to separate the layer height of outlines and infill , if the user want the same value , you create a checked box to join them , but if he wants thick outlines and turbo infill , he can !
Maths:
Speed are the sames whatever the height

lordofhyphens, 2018-01-09:
@studiodyne you should make a feature request on the Slic3r tracker, not talk about it here.

I am unsubscribing from this issue now, so I will not see further posts.

platsch, 2018-01-12:
@studiodyne: @lordofhyphens is right, you should ask for features in the project's tracking system.
Regarding your proposals: you are describing the combined infill feature (or microlayering) which is implemented in slic3r for years. Just set `Combine infill every` to a number > 1.
That should even work with variable layer heights, so you can easily combine both approaches.

As a side note: combining multiple infill layers is not trivial since they generally don't cover the same area. And you want the infill height to be a multiple of the perimeter height.