MarlinFirmware/Marlin #7599: Display error with G28 -

Xserxes, 2017-09-02:
### Bug Report
Marlin 1.1.5
- Description: When using a Z-Probe and Homing with G28 the text is not correct. Function is perfect!
"echo:endstops hit:  **Z:1.00**" schould be "echo:endstops hit:  **Z:-1.00**" -> the minus is missing
- Steps to reproduce:
>>> m851 z-1
SENDING:M851 Z-1
echo:Z Offset -1.00
>>> m500
SENDING:M500
echo:Settings Stored (498 bytes; crc 685)
>>> g28
SENDING:G28
echo:endstops hit:  **Z:1.00**

GMagician, 2017-09-02:
Hello,

maybe I'm wrong but messages are not related.

M851 Z-1 set an offset (I think probe offset) while G28 messages is where endstop has been hit

riverratdc, 2017-09-04:
This sounds like the same problem I describe in issue #7081 (https://github.com/MarlinFirmware/Marlin/issues/7081). I've had the "Endstops Z" message throughout the 1.1.x release branch. The last code I can use that doesn't display the error is RC8. The message does appear to be spurious, as the printer works well in all respects with 1.1.5 EXCEPT for the G28 being immediately followed with the "Endstops Z" message.

thinkyhead, 2018-02-09:
Be sure to disable `ENDSTOPS_ALWAYS_ON_DEFAULT`.

closed 2018-02-09 by thinkyhead