MarlinFirmware/Marlin #7600: Last bugfix. Compilation error with new BABYSTEP_ZPROBE_GFX_OVERLAY enabled

Bergerac56, 2017-09-02:
Last bugfix 1.1.x (Latest commit 386b74f 7 hours ago)
------------------------------------------------------------
### Compile Error
When I compile with `#define BABYSTEP_ZPROBE_GFX_OVERLAY` I get an error: 'ccw_bmp' was not declared in this scope
```
Paste_the_error_text_here
exit status 1
'ccw_bmp' was not declared in this scope
```

------------------------------------------------------------

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1272142/Configuration.zip)

GMagician, 2017-09-02:
Hello,

I tryed your configuration and I get and errore related to SHOW_CUSTOM_BOOTSCREEN option and missing "_booo*.h" include.
I removed sub option to compile but I get an error about cw_bmp, not ccw_bmp.

I made a [PR] to fix it

Bergerac56, 2017-09-04:
@GMagician 
I forgot the Bootscreen.h file in the zip. Sorry for that.

I tried the fix. It works like a charm. Thanks

closed 2017-09-04 by Bergerac56