MarlinFirmware/Marlin #7605: Impossible to load bed leveling mesh without going through calibration

lf-, 2017-09-03:
### Bug Report
- Description: The code spat out from M503 for bed leveling doesn't run because the M420 S1 fails
- Expected behaviour: The bed leveling mesh is loaded and G29 S0 prints this:
```gcode
>>> g29 s0
SENDING:G29 S0
State: On
Num X,Y: 3,3
Z offset: 0.00000
Measured points:
        0        1        2
 0 -0.15000 +0.15000 +0.40000
 1 -0.30000 +0.00000 +0.00000
 2 -0.40000 -0.10000 +0.25000
```
- Actual behaviour: The bed leveling mesh fails to load and G29 S0 states that `Mesh bed leveling has no data.`
- Steps to reproduce:
  - Send example M503 output:
```gcode
M420 S1 Z0.00
G29 S3 X1 Y1 Z-0.15000
G29 S3 X2 Y1 Z0.15000
G29 S3 X3 Y1 Z0.40000
G29 S3 X1 Y2 Z-0.30000
G29 S3 X2 Y2 Z-0.00000
G29 S3 X3 Y2 Z0.30000
G29 S3 X1 Y3 Z-0.40000
G29 S3 X2 Y3 Z-0.10000
G29 S3 X3 Y3 Z0.25000
```
  - M420 will fail with an undescriptive message because `mbl.set_has_mesh(true);` hasn't been called, since it seems to only ever be called when the bed is levelled manually. This hypothesis has been confirmed by adding a test function that calls that, which causes M420 to succeed.

Config files on my fork:

https://github.com/lf-/Marlin/blob/1.1.x/Marlin/Configuration.h  
https://github.com/lf-/Marlin/blob/1.1.x/Marlin/Configuration_adv.h

```gcode
>>> m111 s32 ; in case it actually does something
SENDING:M111 S32
echo:DEBUG:LEVELING
>>> M420 S1 Z0.00
SENDING:M420 S1 Z0.00
Error:Failed to enable Bed Leveling
[ERROR] Error:Failed to enable Bed Leveling

echo:Bed Leveling Off
echo:Fade Height Off
>>> G29 S3 X1 Y1 Z-0.15000
SENDING:G29 S3 X1 Y1 Z-0.15000
>>> G29 S3 X2 Y1 Z0.15000
SENDING:G29 S3 X2 Y1 Z0.15000
>>> G29 S3 X3 Y1 Z0.40000
SENDING:G29 S3 X3 Y1 Z0.40000
>>> G29 S3 X1 Y2 Z-0.30000
SENDING:G29 S3 X1 Y2 Z-0.30000
>>> G29 S3 X2 Y2 Z-0.00000
SENDING:G29 S3 X2 Y2 Z-0.00000
>>> G29 S3 X3 Y2 Z0.30000
SENDING:G29 S3 X3 Y2 Z0.30000
>>> G29 S3 X1 Y3 Z-0.40000
SENDING:G29 S3 X1 Y3 Z-0.40000
>>> G29 S3 X2 Y3 Z-0.10000
SENDING:G29 S3 X2 Y3 Z-0.10000
>>> G29 S3 X3 Y3 Z0.25000
SENDING:G29 S3 X3 Y3 Z0.25000
>>> G29 s0
SENDING:G29 S0
Mesh bed leveling has no data.
>>> m420 s1
SENDING:M420 S1
Error:Failed to enable Bed Leveling
[ERROR] Error:Failed to enable Bed Leveling

echo:Bed Leveling Off
echo:Fade Height Off
```

lf-, 2017-09-03:
Regarding the above commit: it probably shouldn't be merged. The correct solution is likely to verify that there is actually a mesh or kill the check entirely since M503 seems to display a perfectly sane mesh of all zeroes. Another alternative (which makes a lot more sense) is to just not have M420 S1 fail if the leveling is disabled (enabling it is M420's job...).

lf-, 2017-09-03:
Side bug I just exposed: M503 reports `M420 S1` yet `G29 S0` states that leveling is off:

```gcode
>>> M502
SENDING:M502
echo:Hardcoded Default Settings Loaded
>>> M500
SENDING:M500
echo:Settings Stored (498 bytes; crc 45598)
>>> M503
SENDING:M503
echo:  G21    ; Units in mm
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.19 Y80.19 Z800.00 E418.50
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P500.00 R500.00 T500.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Mesh Bed Leveling:
echo:  M420 S0 Z0.00
echo:  G29 S3 X1 Y1 Z0.00000
echo:  G29 S3 X2 Y1 Z0.00000
echo:  G29 S3 X3 Y1 Z0.00000
echo:  G29 S3 X1 Y2 Z0.00000
echo:  G29 S3 X2 Y2 Z0.00000
echo:  G29 S3 X3 Y2 Z0.00000
echo:  G29 S3 X1 Y3 Z0.00000
echo:  G29 S3 X2 Y3 Z0.00000
echo:  G29 S3 X3 Y3 Z0.00000
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P21.16 I1.50 D74.55
>>> G29 S6
SENDING:G29 S6
>>> M503
SENDING:M503
echo:  G21    ; Units in mm
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.19 Y80.19 Z800.00 E418.50
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P500.00 R500.00 T500.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Mesh Bed Leveling:
echo:  M420 S1 Z0.00
echo:  G29 S3 X1 Y1 Z0.00000
echo:  G29 S3 X2 Y1 Z0.00000
echo:  G29 S3 X3 Y1 Z0.00000
echo:  G29 S3 X1 Y2 Z0.00000
echo:  G29 S3 X2 Y2 Z0.00000
echo:  G29 S3 X3 Y2 Z0.00000
echo:  G29 S3 X1 Y3 Z0.00000
echo:  G29 S3 X2 Y3 Z0.00000
echo:  G29 S3 X3 Y3 Z0.00000
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P21.16 I1.50 D74.55
>>> G29 S0
SENDING:G29 S0
State: Off
Num X,Y: 3,3
Z offset: 0.00000
Measured points:
        0        1        2
 0 +0.00000 +0.00000 +0.00000
 1 +0.00000 +0.00000 +0.00000
 2 +0.00000 +0.00000 +0.00000
>>> M503
SENDING:M503
echo:  G21    ; Units in mm
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.19 Y80.19 Z800.00 E418.50
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P500.00 R500.00 T500.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Mesh Bed Leveling:
echo:  M420 S1 Z0.00
echo:  G29 S3 X1 Y1 Z0.00000
echo:  G29 S3 X2 Y1 Z0.00000
echo:  G29 S3 X3 Y1 Z0.00000
echo:  G29 S3 X1 Y2 Z0.00000
echo:  G29 S3 X2 Y2 Z0.00000
echo:  G29 S3 X3 Y2 Z0.00000
echo:  G29 S3 X1 Y3 Z0.00000
echo:  G29 S3 X2 Y3 Z0.00000
echo:  G29 S3 X3 Y3 Z0.00000
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P21.16 I1.50 D74.55
>>> M420 S1
SENDING:M420 S1
echo:Bed Leveling On
echo:Fade Height Off
>>> G29 S0
SENDING:G29 S0
State: On
Num X,Y: 3,3
Z offset: 0.00000
Measured points:
        0        1        2
 0 +0.00000 +0.00000 +0.00000
 1 +0.00000 +0.00000 +0.00000
 2 +0.00000 +0.00000 +0.00000
```

Roxy-3D, 2017-09-03:
We used to have a Bi-Linear label we could put on issues to help with their resolution.   Maybe that explains why there are so many Bi-Linear issues now?  We don't have any way to flag them and get people with similar issues to band together and suggest solutions.  

Maybe the first step is to figure out how to re-create a Bi-Linear label and flag this thread as such?

lf-, 2017-09-03:
I don't think this is bilinear. This is G29 manual mesh leveling.

thinkyhead, 2018-02-09:
Presuming solved.

closed 2018-02-09 by thinkyhead

lf-, 2018-02-11:
Ref: #9088, #9094

ryan11lb426, 2018-02-19:
tbh having the same issue and having no luck even after loading bugfix

thinkyhead, 2018-02-19:
> Having the same issue and having no luck even after loading bugfix. m420 s1 will just immediately disconnect me from my printer.

You should consider posting an issue about that. With details.

derdigge, 2018-07-14:
This issue is still active.
Me MKS Gen 1.4 with Z Probe on zMin and safe_homing active.

```
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Mesh Bed Leveling:
echo:  M420 S0 Z0.00
echo:  G29 S3 X1 Y1 Z-0.15000
echo:  G29 S3 X2 Y1 Z0.00000
echo:  G29 S3 X3 Y1 Z0.00000
echo:  G29 S3 X1 Y2 Z0.00000
echo:  G29 S3 X2 Y2 Z0.00000
echo:  G29 S3 X3 Y2 Z0.00000
echo:  G29 S3 X1 Y3 Z0.00000
echo:  G29 S3 X2 Y3 Z0.00000
echo:  G29 S3 X3 Y3 Z0.00000
```

```
Error:Failed to enable Bed Leveling
echo:Bed Leveling Aus
echo:Fade Height Aus

```

How to fix this?