MarlinFirmware/Marlin #7608: [FR] Print cooling fan PWM to be based on hot-end PWM

nealpalmer, 2017-09-04:
### Feature Request

Marlin has an error "Thermal runaway" that can be triggered by having the print cooling fan blowing too strong.  If the fan is blowing too much, and the hot-end is at 128/128 PWM (i.e. 100%) for too long, then Marlin will cancel mid-print.

I have an environment where the ambient temperature has wild swings daily.  And I would like to be able to use the cooling fan to help with bridging and stringing all the time.  But I can't reliably do this because I get some "Thermal runaway" errors.

What I propose is having an option to make the cooling fan have a PID control over it to try to keep the hot-end at a certain PWM number (settable by the user in the config file).  This would mean that you could have the cooling fan OFF during initial heat-up (while the hot-end is at 100%), and then for the rest of the print, the cooling fan would never run so fast to over-cool the hot-end.

Basically the feedback control for the fan's PWM setting would be the PWM number from the hot-end:
pid_delta = hot_end_pwm - config_setting_fan_desired_hotendpwm;

JerryDurand, 2017-09-04:
If I understand you correctly, you can already do this in both 
Simplify3D and Cura 2.  In Cura you have to make the option visible, in 
S3D you simply add more set points for the fan.  I generally start out 
at 0% for the first layer and ramp up the speed over the next few layers.

If you turn the fan on too fast (say 0 to 100% in one layer) you might 
cool the print head down too much until the PID can catch up.

I use that on my "not-a-Prusa i4" (Sunhokey) and my delta before I sold it.

Unfortunately my two Afinia printers only have one fan, it cools the 
extruder and the print at the same time.  No options there other than 
mechanically blocking it.

nealpalmer, 2017-09-07:
Jerry, you misunderstand.   If I set my part cooling fan to 50% PWM, it will cool my hot-end too much - on some prints (depending on the ambient temperature).  So I can't use a static setting for the fan speed.

I want the fan speed to be automatically as high as possible (i.e. until the hot-end heater PWM gets to a certain %).  This should allow for the maximum amount of cooling for bridging and small features.

The slicing setting for fan PWM would probably still be necessary for the first layer(s).  And the slicing fan PWM would be a useful maximum speed setting for the fan.

hg42, 2017-12-01:
I think, you should probably isolate your heater block, which is a better solution than working around this with software and keep the real problem.

thinkyhead, 2018-02-09:
Closing as a duplicate.

closed 2018-02-09 by thinkyhead