MarlinFirmware/Marlin #7609: Manual bed leveling / bed Z not working

stema84, 2017-09-04:
I have standard Tevo Tarantula, Marlin 1.1.4, started out with an easy config, only changed the direction of Y and margins of bed leveling. Did a manual bed leveling, made sure the Z endstop lets the nozzle to hit the bed a bit, thus I needed positive values to fit a copier paper under the nozzle. Calibrated it, started printing and it pushes the nozzle against the bed, no filament is coming out and the extruder of course is clicking.
I've tried Tuning it with "Bed Z", not knowing if it should be a +/- value, I tried both while still printing the first layer (again, I don't know if it should work at that point but since temperature changes it accepts, I would assume it should). Nothing happened, I set it from +1.000 and to -1.000, no difference. Bed leveling I did with the bed heated up to 110C, and I have a glass on it, it didn't even cool down between the leveling and the printing.

karabas, 2017-11-27:
The same with 1.1.6

thinkyhead, 2018-02-09:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch in case the issue is already fixed. If we already got it then you can close the issue. If you still see the bad behavior then we can investigate further.

boelle, 2019-02-19:
@stema84 is the problem gone with latest bugfix 2.0 ?

stemaDev, 2019-02-19:
I cannot test it since I'm continuously printing, but I had 1.1.9 release installed, not the 1.1.x bugfix, so it could be that it's been fixed since then. I also got the info, that there are many jumpers on the board originally, and all should be removed because it can cause such an issue (I thought they are on pins which are not in use)

boelle, 2019-02-19:
you need to test so that we can know if the problem is gone

stemaDev, 2019-02-19:
I will, but I don't know when, so I will close this issue for now and may reopen it if I have the chance, thanks for the help

closed 2019-02-20 by thinkyhead

DrywFiltiarn, 2019-02-26:
I have a similar issue running Marlin-2.0.x on an Ender-3. The "Bed Z" setting does not seem to do anything either towards positive or negative values. When using babystepping during printing of a initial primeline (using cura) I can correct to the correct offset value. Setting "Bed Z" to -0.08 does nothing, whereas babystepping to -0.08 get's the nozzle at the right offset height.

stema84, 2019-02-27:
Oh god, sorry for the confusion.. I opened an issue just 2 weeks ago and I totally thought that this is that so please disregard my comments from this month.

thinkyhead, 2019-03-03:
> Setting "Bed Z" to -0.08 does nothing, whereas babystepping to -0.08 [gets] the nozzle at the right offset height.

Do you use your probe for homing Z?
Are you using `G29` and saving the mesh with `M500`?

This should do the trick for you:
```gcode
M851 Z-0.08
M500
G28
G29
M500
```

mbosgraaf-wirelab, 2019-03-03:
I’m not using a probe at all, leveling my glass bed with a feelergauge (0.08mm) so I expected (perhaps falsely) that setting Bed Z would result in effectively actually putting Z at 0 (0.08 nozzle offset + -0.08 Bed Z = Z0).

At the moment I just do the z offset through the slizer, but ideally I would do this on the machine itself to be able to adjust it on the go if needed without needing to reslice with a different z-offset.

Babystepping does allow for this, but at the disadvantage having to do it at every start of a print.

thinkyhead, 2019-03-03:
> I’m not using a probe at all…

We know. `MESH_BED_LEVELING` is the first clue.

> …at the disadvantage having to do it at every start of a print…

You can use `M206 Z` to set an offset to your Z home position. If you find after homing that the nozzle is 2mm above the bed, then you would use `M206 Z2` followed by `M500` to ensure that the machine applies that correction after `G28`.

mbosgraaf-wirelab, 2019-03-03:
M206 sounds indeed like what I would expect Bed Z would also relate to. But as I understand from above responses these are different things right? Curious now, what the Bed Z setting actually does apply to then if it’s not a Z offset 🤔

thinkyhead, 2019-03-03:
The "Bed Z" is an element of the `MESH_BED_LEVELING` feature. It's an additional factor applied to the Z position during printing when leveling is enabled. If the "Bed Z" value is negative the nozzle will be moved closer to the bed.

mbosgraaf-wirelab, 2019-03-03:
Curious case then as now I’m wondering what it takes then to activate mesh bed leveling (which was enabled in config), running the guide for setting it up was done (with result 0.0 on all 9 points checked as the glass bed was already trammed with a feeler guide at 0.08mm), yet Bed Z neither a positive or negative value changed anything.

thinkyhead, 2019-03-03:
When in doubt, `M420 S1` will always enable leveling, and `M420` will tell you the current state.

mbosgraaf-wirelab, 2019-03-03:
Thank you, will look into it further later :) atm I have mesh leveling disabled again to save room for other features (darn melzi boards are low on storage).

thinkyhead, 2019-03-06:
True. The most economical leveling option is `MESH_BED_LEVELING` but even it uses a KB or two.