MarlinFirmware/Marlin #7610: new to 3d printing please help

blakeable, 2017-09-04:
I pasted my error below I'm new to 3d printing I'm trying to download firmwear  

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```
Arduino: 1.8.4 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\blakewock\AppData\Local\Temp\Temp1_Anycubic kossel_20170712.zip\Anycubic_kossel\Anycubic_kossel\Anycubic_kossel\Anycubic_kossel.ino:37:26: fatal error: MarlinConfig.h: No such file or directory

 #include "MarlinConfig.h"

                          ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

Scencea, 2017-09-04:
It just seems that you don't have the required file in your folder.

Extract everything again into one folder, open Marlin.ino and try to upload it.

blakeable, 2017-09-04:
sorry but what is marlin.ino

Scencea, 2017-09-04:
Extract the whole folder -> doubleclick on Marlin.ino ->the whole project opens-> edit configuration.h (or paste it in the folder) -> strg+u ->finished

blakeable, 2017-09-04:
I'm not sure what the strg+u means? I re extracted everything opened marlin.ino  and pasted the configuration file in its still having the same error

Tannoo, 2017-09-05:
Wouldn't it be easier to google "How to setup Marlin firmware"?

The easiest way is to make sure you have the Arduino IDE installed.
Then extract the marlin firmware to it's own directory.
Open the Arduino IDE and click file > open.
Browse to the folder you extracted Marlin into and select the `Marlin.ino` file.
From there, click the tab at the top `configuration.h` and start setting the options needed for your printer.
Ensure you have at least a USB connection from your PC to your printer.
When you are done changing the configuration (youtube tom's guide for detailed info), click the upload arrow.

If you have troubles with the IDE.... go here > www.arduino.cc

cdedwards, 2017-09-05:
what would be really nice, is a guide to keeping a source up to date with git while keeping your own settings in the configuration.h and configuration_adv.h files. ROXY_3D posted a message to me for creating a repo and making changes to the source so I could do a pull. But of course I've misplaced it. it did not however allow what I'm looking for..

Tannoo, 2017-09-05:
If you fork Marlin, then clone it to your desktop, you can then change the config files to suit you. Then.. you can update your cloned copy of your fork (the fork will be automatically updated if anything in Marlin changes).
Updating your clone just updates the changes.

benlye, 2017-09-05:
One important extra step - best practice is to create a new branch for your configuration so that you are not applying your changes directly to the default branch of your fork.  That way you keep a pristine branch you can use as the basis for any pull requests.

E.g. in my fork I have [bugfix-1.1.x](https://github.com/benlye/Marlin/tree/bugfix-1.1.x/) which is exactly the same as MarlinFirmware/bugfix-1.1.x.

I use these commands in the Git shell to resync it with Marlin from time to time:
```
git fetch MarlinFirmware
git checkout bugfix-1.1.x
git reset --hard MarlinFirmware/bugfix-1.1.x 
git push origin bugfix-1.1.x --force 
```
I also have the branch where I keep my configured version:
https://github.com/benlye/Marlin/tree/benlye-ramps

Which is also up to date with bugfix-1.1.x.  I keep it up to date using the 'Update' button in Github Desktop, with 'MarlinFirmware/bugfix-1.1.x' selected as the origin.

Tannoo, 2017-09-05:
Yes....I forgot to mention that part.

blakeable, 2017-09-05:
thank you so much for your help I got it I'm stupid I haven't used a computer in a while

closed 2018-02-09 by thinkyhead