MarlinFirmware/Marlin #7612: incorrect errormessage

ctday, 2017-09-05:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
Arduino: 1.8.3 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\MarlinConfig.h:39:0,

                 from D:\cygwin64\home\Scots\Marlin\Marlin\Marlin.ino:31:

SanityCheck.h:973: error: #error "TEMP_SENSOR_1 shouldn't be set with only 1 extruder."

   #error "TEMP_SENSOR_1 shouldn't be set with only 1 extruder."

    ^

exit status 1
#error "TEMP_SENSOR_1 shouldn't be set with only 1 extruder."

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
------------------------------------------------------------_ 
I had CONFIGURE.H set for two extruders, one nozzle, and two hotend temp sensors. This is indeed an error, but the message should read "...with only 1 nozzle."

Tannoo, 2017-09-05:
Please rename your `configuration.h` file to `configuration.txt` and post it here.

ctday, 2017-09-05:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1277996/Configuration.txt)

Tannoo, 2017-09-05:
Why don't you have `MIXING_EXTRUDER` enabled?

ctday, 2017-09-06:
By my understanding, neither MM nor Prometheus is a mixing system, i.e., one where multiple filaments are fed to the nozzle simultaneously. They each withdraw one filament before feeding and clearing another filament. Also, that configuration is not necessarily a complete, consistent configuration; I've just started.

Tannoo, 2017-09-06:
Ok... multiples and mixing is out of my league just yet.
Someone else will have to step in. Sorry

ctday, 2017-09-06:
No problem.

GMagician, 2017-09-07:
Sorry but if you have SINGLE_NOZZLE way do you have 2 temp sensors configured?

GMagician, 2017-09-07:
Changed messages in sanitycheck from extruder to hotends, since these should be correct error messages.
I'll post a PR

ctday, 2017-09-07:
The two temperature sensors and one nozzle was a mistake on my part, but it triggered the incorrect message.

GMagician, 2017-11-18:
I think this can be closed

closed 2018-02-09 by thinkyhead

MohamedSheblSaQr, 2020-04-16:
i have the same error now while trying edit  CONFIGURE.H set for two extruders one nozzle for each extruder. it's my graduation project and i need help plz.

thinkyhead, 2020-04-17:
@MohamedSheblSaQr — 
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y