MarlinFirmware/Marlin #7613: [FR]Whenever marlin received a msg on halt state, it could reply an error msg for the halt reason.

miblooming, 2017-09-05:
------------------------------------------------------------
### Feature Request
[Request]
Whenever marlin received a msg on halt state, it could reply an error msg for the halt reason.

[Phenomenon]
When marlin is halt,there's no msg back for host on the serial port. And the host can not distinguish Marlin-Halt from Not-Connected
such as:
Error:MAXTEMP triggered, system stopped! Heater_ID x*
Error:MINTEMP triggered, system stopped! Heater_ID:x*
Error:Pritner stopped and use M999 to restart it .
------------------------------------------------------------

Bob-the-Kuhn, 2017-09-05:
As best I can tell Marlin already prints an explanation as to why the kill/halt routine is called. 

If you can provide details on how to re-produce a halt that doesn't print an explanation I'll be glad to investigate further.

miblooming, 2017-09-06:
I'll sorry for my bad expression.
My mean is "After" marlin halt,there's no msg back for host on the serial port.
Case 1:
If a host (such as a wifi module or a LCD )connected to marlin by UART by AUX-1 in Ramps1.4,
When the Mega2560 power on, marlin and host is start up at the same time, 
Malrin detects TEMP ERROR, and send out a msg，
but the Host may not finish Initialization，then the msg is lost for the host.

Case 2:
After marlin halt, the users plug the host to Ramps1.4, the host would not get msg for marlin.(Many 
 DIYer do like this, although it is not good to hot plug the host)

Bob-the-Kuhn, 2017-09-06:
I think that you want the printer to periodically send  a "I halted because ..." message.

miblooming, 2017-09-07:
The printer sending msg proactively, or be triggered by received msg after halt, is both acceptable.
The key point is the Host having a way to know what happened on the printer when marlin halt.

thinkyhead, 2018-02-09:
See `ACTION_ON_KILL`.

closed 2018-02-09 by thinkyhead