MarlinFirmware/Marlin #7616: Disconnecting thermistor no longer triggering MINTEMP error and lockout

marcio-ao, 2017-09-05:
- Description: Disconnecting thermistor no longer triggering MINTEMP lockout
- Expected behaviour: Upon disconnecting the thermistor, the temperature drops below HEATER_0_MINTEMP and this should cause an error and printer lockout.
- Actual behaviour: Nothing appears to happen when the thermistor is disconnected.
- Steps to reproduce:
  - Disconnect thermistor and/or print head.

marcio-ao, 2017-09-05:
I noticed there is a check [here](https://github.com/MarlinFirmware/Marlin/blob/9c00cff89836706776f6353635368a47cc5bf881/Marlin/temperature.cpp#L718), but it only seems to be active when HEATER_0_USES_MAX6675 is enabled.

I tried making a similar check for when that isn't defined:

  #if ENABLED(HEATER_0_USES_MAX6675)
    if (current_temperature[0] > min(HEATER_0_MAXTEMP, MAX6675_TMAX - 1.0)) max_temp_error(0);
    if (current_temperature[0] < max(HEATER_0_MINTEMP, MAX6675_TMIN + .01)) min_temp_error(0);
  #elif ENABLED(LULZBOT_MINTEMP_WORKAROUND)
    if (current_temperature[0] > HEATER_0_MAXTEMP) max_temp_error(0);
    if (current_temperature[0] < HEATER_0_MINTEMP) min_temp_error(0);
  #endif

The problem is that the printer mintemps immediately at startup, even when the thermistor is connected. I don't understand why.

JerryDurand, 2017-09-05:
I haven't looked at the code there but most probably the system hasn't 
taken enough readings to fill out its smoothing/averaging routine.  
There has to be a delay at any startup to allow the analog inputs and 
their digital filters to stabilize.

marcio-ao, 2017-09-11:
@JerryDurand: You were correct. I managed to fix the problem using the following hack:

    static int delayBeforeStartMeasuring = OVERSAMPLENR; \
        if(delayBeforeStartMeasuring > 0) { \
            delayBeforeStartMeasuring--; \
        } else { \
            if (current_temperature[e] > HEATER_0_MAXTEMP) max_temp_error(0); \
            if (current_temperature[e] < HEATER_0_MINTEMP) min_temp_error(0); \
        }

I still feel like this is sort of a hack and there ought to be a better way, but for now this solves my issue.

thinkyhead, 2018-02-09:
Marlin no longer throws temp errors until you try to enable a heater.

closed 2018-02-09 by thinkyhead