MarlinFirmware/Marlin #7617: babystep_zprobe_offset not changing during print

empakoso, 2017-09-05:
### Bug Report

- Description: ---
Vrs1.15 BugRelease Downloaded 09-03-2017
When babystepping is defined with Zprobe Offset or M851 it does not change the nozzle height during printing.  
This may or may not be related, but after defining:
line607  #define BABYSTEP_ZPROBE_OFFSET
The control menu displays Z-offset value in a sub menu.  Where as, when line607 is commented out, the Z-offset is displayed without having to select it.

- Expected behaviour: ---
When line607 is commented out, the babystepping works as expected by moving the nozzle closer or further away from the bed during the first layer print.
When line607 is activated and tied to M851, the babystepping value changes on the LCD display but the nozzle does not move.

tcm0116, 2017-09-07:
Are you using bilinear bed leveling?

empakoso, 2017-09-07:
Ever since I switched over to UBL

empakoso, 2017-09-10:
I've completed this using M851 Z(value) codes.  I did not realize that when this value is edited the Z axis moves to that compensation point immediately after this code execution. How convenient.
 To help out anyone else in this same situation (determining the perfect Z offset) here are my steps to find a perfect Z offset:
; heat bed and hot end for at least 5min to ensure dimensional stability at working temps.
G28 ; home using probe
; because we home at an offset to the nozzle we have to remember that the bed may ascend or descend slightly due to imperfections.
; meaning you must move the nozzle to the probing location before adjusting z offset.
; for cartesian it is the centre point between the given Bed (X,Y) dimensions in the v1.1.5 fw
; mine is Cartesian therefore
G1 Z5
G1 X100 Y100 F4000
; my shim is a piece of paper, measure thickness using a caliper is 0.08mm
; therefore I will move the hot end down to this height
; I place my paper under the hot end then execute the following:
G1 Z0.08
; paper should just slightly grab under this position height if so your good, else continue
; you could move your z axis manually using your LCD to get a quick idea of how much to adjust your nozzle (if you need to go down, there are a couple of steps needed to go beyond you endstop, I'll let you figure that out)
; editing the perfect Z offset
; send nozzle up 5mm
G1 Z5
; now adjust your offset
M851 ; to 
; too get closer to the bed - reduce by going with a lesser number try increments of 0.05mm
M851 Z(new value)
; the z axis will shift - you can hear the motors shift.
; now bring the z axis back down to the paper
G1 Z0.08
; if it grab slightly, you are good, otherwise repeat the above:
G1 Z5
M851 Z(new value)
G1 Z0.09
; if it grabs slightly you are done, else repeat.

My final adjustment is to measure my G26 UBL mesh layer.  I will adjust the M851 to achieve the 0.2mm thickness.
My absolute final adjustment is by measuring the first layer ring around my print.  I add or subtract this adjustment in my slicer. (It's different and I don't really know why)
I like this method because you can determine by trial and error achieving a a value in the hundredths of a mm. I didn't need to go beyond a resolution of 0.05mm increments.  It's also a good thing to figure out equal steps per increment by using you steps/mm value and dividing it up to the nearest 0.05 increment as a whole step value.  I found my resolution was a bit more predictable using a 0.04mm in my case.
Just my $0.02

closed 2017-09-10 by empakoso