MarlinFirmware/Marlin #7618: UBL relation to M851 Zprobe Offset - Please help to confirm

empakoso, 2017-09-05:
Need Assistance confirming
Vrs1.1.5 BugFix - Downloaded 09-03-2017
This problem was experienced  when trying to find the perfect Zprobe Offset with UBL.  This procedure is what was used to make adjustments until the correct offset was estimated.  The paper method (0.07mm shim) could not be used reliably (accurately-precisely possible mechanical back lash).

- Description: UBL activated.  Tried to determine exact Zprobe offset so that a perfect layer thickness is printed on the first layer.  Just changing from M851 Z-0.09 to M851 Z-0.45 did not affect the next print.  The changes were edited using all three methods M851, LCD, and by uploading the Firmware(M502 & M501 & M500 using a saved mesh).
- Expected behaviour: Change Zprobe offset = change on the next print.  The goal is to print a 0.2mm layer and have it measure 0.2mm.
- Actual behaviour: After changing the Zoffset either by M851, LCD, or FW updated, the changed Zprobe offset would display in the Control menu.  The print would still show the previous thickness.

The G29 W command was given each time.  The Zoffset value that it was originally probed with would be reported.
Then it was realized that a G29 P1 and G29 P3 was needed so that the new Zprobe offset would be reported as the Z value.

Just to confirm, these are the steps taken:
Change Zprobe offset
G29 L0 ; load previously saved mesh to check
G29 A ; activate UBL
G29 W ; check UBL values
heat and wait for 5minutes Bed & Nozzle
G29 P1 ; create mesh
G29 P3 T1 ; check points (there will be NAN values)
G29 P3 T1 ; check points (ensure NAN values are extrapolated and not present)
G29 F20 ; set fade height
G29 A ; activate UBL
G29 W ; check UBL values
G29 S0 ; save the mesh
begin printing with new Zprobe offset
G26 ; test print of mesh
check for a relative 0.02mm layer thickness if not change Zprobe offset
if an odd mesh point is out of wack, perform G29 P2 to correct that point
G29 S0

Or would using G29 P6 C?.??mm to offset the mesh to the correct M851 value be faster by eliminating the need to do the G29 P1 all over again?

Knowing a little more about the above steps would likely allow the use of the M851 Z0, G28, G92 Z0, M114, M851 Z(from M114) method to determine Zprobe offset.  If I have to adjust the mesh after an M851 adjustment then this would explain why it wasn't working in the first place.

Side Note: The Zprobe has been changed to an IR sensor to eliminate the following conditions; mechanical backlash of sensor deployment, reflectivity, thermal, and electrical variances caused by other types of probes.  The M48 reports a 0.001 stdev using the Mini Differential IR Height Sensor by Filastruder.

Roxy-3D, 2017-09-05:
When you change the Z_PROBE_OFFSET_FROM_EXTRUDER, are you saving the new value in EEPROM?  Because the next time the printer is reset or comes alive from power up, it will load that value from EEPROM.

empakoso, 2017-09-05:
Yes I am.  I forgot to state G29 S0 then M500 at the end of my steps.

Roxy-3D, 2017-09-05:
I think you should disable UBL.  Turn it off.    And see if  you can print a small object in the center of the bed.   If you can do that, it will be much easier to bring UBL up.   And if you can't do that...  It will be much easier to figure out what has to change.

empakoso, 2017-09-06:
Well that makes a complicated issue nice and simple! Thank you.

closed 2017-09-06 by empakoso

empakoso, 2017-09-06:
Wow! One very successful UBL setup.  Thanks for the setup tip.  My magic number is 
M851 Z-0.260

empakoso, 2017-09-07:
Happy UBL user!  Thanks To @Roxy-3D for all the hard work.
[image]

Roxy-3D, 2017-09-07:
Wow!  You must have a very big diameter nozzle.  .5mm ????    But the point is, you can see you are getting good first layer adhesion across 100% of your bed!   And it should stay that way if always put the glass back down the same way.  (Is that klapton tape?   If so...  it might vary slightly each time you replace the tape and doing another G26 with a G29 P4 R editing session will let you zero in again very quickly.)

empakoso, 2017-09-07:
Yes 100% adhesion is the sole purpose and what a difference in overall print finish when the first layer isn't all messy.
It's funny you say that since I helped someone with the same nozzle and observed the same thing.  Guess I am buying a new nozzle.
Yes it's Kapton tape. Thanks for the advise on the tapes variance, never crossed my mind.
Thanks again. Your work hasn't gone without much appreciation.

Roxy-3D, 2017-09-07:
> Yes it's Kapton tape. Thanks for the advise on the tapes variance, never crossed my mind.
Thanks again. Your work hasn't gone without much appreciation.

When you put down a new layer of klapton tape...   Don't regenerate the mesh.   The one you have will be 'close'.  Just do a G26 to generate a pattern...  And edit any areas of the mesh that are not identical to the previous layer of tape.   It should just take 5 minutes to print the validation pattern....  And another 5 minutes to edit any areas that changed...

empakoso, 2017-09-07:
Great! Will do.

empakoso, 2017-09-08:
Quick question, RE: G29 P4 
Is the intention that the probe makes a true signal before the encoder knob acknowledges the reading?
On this note I have learned to visualize the mesh output values.  Do they represent the measured difference from a point?

Roxy-3D, 2017-09-08:
At 80,000'....  The mesh point values in all of the mesh bed leveling systems are this:   

They are the compensation value needed at that position on the bed so the nozzle is at the 'correct' height.

A lot of further interpolation is done to correct for the nozzle height at various positions within the 'mesh cell'.  Almost never is the nozzle at a 'mesh point'.   Some bi-linear correction is usually done to approximate the 'correct' compensation at various locations within a mesh cell.

empakoso, 2017-09-08:
You are absolutely correct in that it took 3 attempts of G26 to perfect the mesh. UBL is fast and simple.
Being my first attempt in enabling UBL, I was intent on knowing all there is about all the features of UBL. The learning curve has really been satisfying.  Now mesh setup is so fast (not so fast when I was stumbling through the new subset of codes.
What is the relationship between:
G29 P6 C0.04 ; mesh shift up by 0.04mm
And
M851 Z-0.22 ; (from -0.26) smaller offset, bigger gap
Are both needed or just the one?

empakoso, 2017-09-08:
In the terms of Red Foreman: "I'm a dumbass!"
Just figured out G29 P4 (H is defaulted to z clearance between probes).  I will use G29 P4 H0.09 so that a paper shim begins to grabs! Oh it's a wonderful day ... (I should get out more)

empakoso, 2017-09-08:
Once again thank you @Roxy-3D you've answered my question

G29_UBL.cpp:
 P6 Phase 6 Shift Mesh height. The entire Mesh's height is adjusted by the height specified
* with the C parameter. Being able to adjust the height of a Mesh is useful tool. It
* can be used to compensate for poorly calibrated Z-Probes and other errors. Ideally,
* you should have the Mesh adjusted for a Mean Height of 0.00 and the Z-Probe measuring
* 0.000 at the Z Home location.

empakoso, 2017-09-09:
OK! Final completion! 100% UBL system applied successfully.  Thanks again @Roxy-3D 
You were right it is a 0.05mm nozzle.
[image]

Roxy-3D, 2017-09-09:
It looks like you have one small area to edit...   You are missing the semi-circles on two of those mesh points in the lower left.   And it looks like not enough filament is coming out to connect the mesh points.

But that might be where you started the print and the nozzle wasn't sufficiently primed.  Probably that is the case because the vertical lines from those two mesh points are correct.  So the mesh point heights there are correct.

empakoso, 2017-09-09:
You are right. Just did a larger print and the first layer feathered out a bit in that corner.
To have this kind of control is awesome.
I've even used your G29 S-1 command to export the mesh dimensions in gcode. And just for a laugh I jettisoned the mesh, then 'printed' the mesh points and saved it in slot 0.  Even saved my M851 Z offset all in one convenient gcode file.
I am a true believer. Nice work.

zwep, 2020-01-13:
> OK! Final completion! 100% UBL system applied successfully. Thanks again @Roxy-3D
> You were right it is a 0.05mm nozzle.
> [image]

Quick question, where did you find the .stl file to print this object? Would love to give it a try as well

Vertabreak, 2020-01-13:
that is what G26 looks like when it works correctly, i personally have never gotten it to work that well.

zwep, 2020-01-14:
> that is what G26 looks like when it works correctly, i personally have never gotten it to work that well.

Thanks! Somehow the G26 protocol did not react to my changes in the Z-probe-offset. Might be because I only have Bilinear bed leveling enabled.. but still strange, because the code looks like to be telling something different

empakoso, 2020-01-14:
G29 is only for UBL.  Each crosshairs is a probe location.  The idea is to provide a visual of each probe location.