MarlinFirmware/Marlin #7619: Retraction is too slow

cfelicio, 2017-09-05:
-----------------------------------------------------------
### Bug Report
- Description: When doing a tool switch retraction (e.g. G92E0 G1 E-55 F8000), the speed is very low
- Expected behaviour: If I issue a retraction command with a fast speed like F8000, I expect the retraction to be fast.
- Actual behaviour: No matter how high I go (I went all the way to F10000 before I noticed it was not making a difference), retraction seems to be limited to around F2000, anything higher won't go faster.
- Steps to reproduce:
Issue a retraction command (e.g. G92E0 G1 E-55 F8000) and see if it retracts fast (you can run G92E0 G1 E-55 F2000 as comparison)

I'm attaching my config files, maybe it's a configuration issue. I uploaded repetier and the same retract goes super fast, at the expected speed.

The reason why I need this, is because I want to do filament switching with a Y splitter, and if it's not fast enough, it will form a string and jam.

[OneDrive-2017-09-05.zip](https://github.com/MarlinFirmware/Marlin/files/1278433/OneDrive-2017-09-05.zip)

KevMags, 2017-09-05:
configuration.h (latest bug fix) line #525

#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 25 }

Maximum federate for E is 25 mm/s by default.  Raise value as needed.

cfelicio, 2017-09-05:
Awesome, thanks!! :-)

cfelicio, 2017-09-05:
I just checked, I had already raised this value. Here is my setting:

#define DEFAULT_MAX_FEEDRATE          { 500, 500, 5, 250, 250 }

Any other ideas?

hexane360, 2017-09-09:
DEFAULT_MAX_FEEDRATE can be overrided with M203. Is that anywhere in your start g-code?

Otherwise, are you sure that's the version of Configuration.h that you've flashed to your printer?

cfelicio, 2017-09-10:
no M203 on the gcode, the same gcode works fast on repetier.

I reflashed Marlin back just to confirm, ensuring the above config file is used, and the printer immediately goes back to the odd behavior, where the extruder does not retract fast.  :(

thinkyhead, 2017-11-14:
Do you see the same issue in current `bugfix-1.1.x` branch?

closed 2017-11-14 by thinkyhead