MarlinFirmware/Marlin #7620: Set the Points for LCD Leveling - ANET A8 example is wrong

SphaeroX, 2017-09-05:
Hi, if I level via the display on my Anet A8 with the example configuration, then I have the problem that it is located left behind the hole for the screw. Where can I adjust this without changing the size of the print bed?

Tannoo, 2017-09-05:
```
  #define LEFT_PROBE_BED_POSITION 15
  #define RIGHT_PROBE_BED_POSITION 170
  #define FRONT_PROBE_BED_POSITION 20
  #define BACK_PROBE_BED_POSITION 170
```

SphaeroX, 2017-09-06:
I have already tried to change these values but unfortunately it has no effect on it.

But I'll try it again tomorrow or so, thank you for your help.

Soulreaver76, 2017-09-06:
I have the same problem
Nothing changes at the position.

Phr3d13, 2017-09-06:
I think you may have to load firmware defaults or load from eeprom or whatever it's called after changing some things.

closed 2018-02-09 by thinkyhead