MarlinFirmware/Marlin #7621: LCD_BED_LEVELING wont work at Anet A8

SphaeroX, 2017-09-05:
I tried to level my print bed with the help of LCD_BED_LEVELING, after I levelled my print bed and saved the settings, I tried to print something.

But when I was printing, I realized that I was too close to the printing bed, so I decided to go up a few millimetres to test it. Marlin's just not hiring.

I also tried the launch code.

G29; bed level 
M420 S1; bed leveling on

Here are the settings I changed for levelling:

#define LCD_BED_LEVELING
#define MESH_BED_LEVELING
#define PROBE_MANUALLY

Tannoo, 2017-09-05:
Marlin is volunteer work.

Could you post your config files?

Soulreaver76, 2017-09-06:
I have the same problem.

SphaeroX, 2017-09-06:
When I connect the printer to my PC via USB and open the Arduino console, I see that the levelling values are stored. But he doesn't use them for printing.

MasterPIC, 2017-09-13:
I found that the home command disables home leveling (you can check the state by G29).
To solve the issue you've to send M420 S1.

I hope it helps.

SphaeroX, 2017-09-14:
Thanks for the solution

closed 2017-09-14 by SphaeroX

Roxy-3D, 2017-09-14:
Not all of the bed leveling systems disregard the users preference and clear it with a G28

https://github.com/cdedwards/Marlin-cde/blob/bugfix-1.1.x/Marlin/Marlin_main.cpp#L3940-L3947

UBL was designed to respect the user's preference and preserve it across a G28.

guiseco, 2017-09-14:
UBL on anet A8 ?

SphaeroX, 2017-09-14:
Yeah UBL wont Work with Anet Boards (memory) ;)