MarlinFirmware/Marlin #7623: New to writing firmware having issues (mentor would be helpful)

bluewheels145, 2017-09-06:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

Tannoo, 2017-09-10:
Well.. Marlin is already written.. just evolving.
Download, extract, compile, upload, enjoy.

No writing involved.

bluewheels145, 2017-09-10:
My bad, I meant editing it to your individual printer, if that makes sense.

Roxy-3D, 2017-09-10:
First...  Change the Mother Board definition to match the controller you are using.   Then start setting up stuff like declaring if you have max or min endstops on each axis.    See if you can move the nozzle using PronterFace.   There are a fair amount of configuration settings to tell the firmware about your printer.

Just start working through them and the printer will come more and more alive.

Tannoo, 2017-09-10:
Those setting are in the `configuration.h` file.

You can use the Arduino IDE to edit, or you can use your favor text editor.

Tannoo, 2017-09-10:
There are other settings also in `configuration_adv.h`.

closed 2017-11-27 by thinkyhead