MarlinFirmware/Marlin #7625: Auto-leveling no longer works in Marlin 1.1.5 for printers with corner washers

marcio-ao, 2017-09-06:
When I try to compile the LulzBot firmware using Marlin 1.1.5, I get the following error:

SanityCheck.h:748:5: error: static assertion failed: ABL_PROBE_PT_1_Y can't be reached by the Z probe.
     static_assert(WITHIN(ABL_PROBE_PT_1_Y, MIN_PROBE_Y, MAX_PROBE_Y), "ABL_PROBE_PT_1_Y can't be reached by the Z probe.");

The issue is that on the LulzBot printers probing happens on corner washers that are outside of the print area (but still within the endstops). Digging further into the code, I see an incorrect assumption is made that the probe points fall within the bed area. This assumption is enforced [here](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Conditionals_post.h#L859):

```
#define MIN_PROBE_X (max(X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER, 0))
      #define MIN_PROBE_Y (max(Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER, 0))
      #define MAX_PROBE_X (min(X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER, X_BED_SIZE))
      #define MAX_PROBE_Y (min(Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER, Y_BED_SIZE))
```

These four assumptions are incorrect on the LulzBot printers or any other printer which has bed leveling points are outside of the print bed area.

This assumption was introduced in commit [51864fd365969337b34f2522fb59ead806674e94](https://github.com/MarlinFirmware/Marlin/commit/51864fd365969337b34f2522fb59ead806674e94)

fiveangle, 2017-09-06:
There was much discussion about this because users with simple cartesians were routinely confused by configuration.  I think the possibility of requiring probing outside of the bed wasn't considered.

I suppose a sanity check warning could be triggered if user set probe points outside of the new `X_BED_SIZE, Y_BED_SIZE` but within `X_MAX_POS, Y_MAX_POS`, and only halt compile if probe points were outside of `X_MAX_POS, Y_MAX_POS`.  It looks like it would be a fairly simple change to Sanitycheck.h but then a warning doesn't do much to protect users if they don't see it (perhaps a multi-line warning bracketted by horizontal rule made of "#############"s top/bottom would be warranted ?)

I began working on a change to default configuration to identify the probe points relative to bed-origin using the new `*_BED_SIZE` values rather than absolute X/Y, pushing the calcs into Conditionals_post.h that would effectively set default probe boundaries on the bed...

For example commenting out this in Configuration.h by default:
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L874-L878
```
  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 15
  #define RIGHT_PROBE_BED_POSITION 170
  #define FRONT_PROBE_BED_POSITION 20
  #define BACK_PROBE_BED_POSITION 170
```
Then if not defined, setting them in Conditionals_post.h as if they were defined in Configuration.h like so (with additional logic to handle the various *_PROBE_OFFSET_FROM_EXTRUDER possibilities and whatever might be needed by delta, scara, etc):
```
  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION (X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER)
  #define RIGHT_PROBE_BED_POSITION (LEFT_PROBE_BED_POSITION + X_BED_SIZE)
  #define FRONT_PROBE_BED_POSITION (Y_MIN_POS + Y_PROBE_OFFSET_FROME_EXTRUDER)
  #define BACK_PROBE_BED_POSITION (FRONT_PROBE_BED_POSITION + Y_BED_SIZE)
```

Then adding an additional comment in the commented out section of Configuration.h example indicating   

```
// Probe boundaries (where the probe can reach) are automatically set based on bed size.
// Uncomment here to manually set probe boundaries (for example, outside of the bed area).
```

But I got busy with life and just haven't gotten back around to it :/  On the plus side, Burninman was awesome ;)

I personally think the above changes would be best but I haven't discussed deeply with anyone yet.

marcio-ao, 2017-09-06:
@fiveangle: In our branch of the firmware I went with the simplest solution I could think of, which was a new conditional PROBE_POINTS_OUTSIDE_OF_BED, which restores the old behavior of "Conditional_post.h" as follows:

```
    // Boundaries for Cartesian probing based on set limits
    #if ENABLED(PROBE_POINTS_OUTSIDE_OF_BED)
      #define MIN_PROBE_X (X_MIN_POS)
      #define MIN_PROBE_Y (Y_MIN_POS)
      #define MAX_PROBE_X (X_MAX_POS)
      #define MAX_PROBE_Y (Y_MAX_POS)
    #elif ENABLED(BED_CENTER_AT_0_0)
      #define MIN_PROBE_X (max(X_PROBE_OFFSET_FROM_EXTRUDER, 0) - (X_BED_SIZE) / 2)
      #define MIN_PROBE_Y (max(Y_PROBE_OFFSET_FROM_EXTRUDER, 0) - (Y_BED_SIZE) / 2)
      #define MAX_PROBE_X (min(X_BED_SIZE + X_PROBE_OFFSET_FROM_EXTRUDER, X_BED_SIZE) - (X_BED_SIZE) / 2)
      #define MAX_PROBE_Y (min(Y_BED_SIZE + Y_PROBE_OFFSET_FROM_EXTRUDER, Y_BED_SIZE) - (Y_BED_SIZE) / 2)
    #else
      #define MIN_PROBE_X (max(X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER, 0))
      #define MIN_PROBE_Y (max(Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER, 0))
      #define MAX_PROBE_X (min(X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER, X_BED_SIZE))
      #define MAX_PROBE_Y (min(Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER, Y_BED_SIZE))
    #endif
```
I don't have a strong preference how it's done, I just wanted to bring it up as an issue so that it could be addressed in whatever way was most appropriate. I think the solution you proposed would be fine.

thinkyhead, 2017-11-27:
Rather than adding (yet another) obscure setting, let's just make these overrideable in the configs. So, in `Conditionals_post.h`…
```cpp
/**
 * Bed Probing rectangular bounds
 * These can be further constrained in code for Delta and SCARA
 */
#if ENABLED(DELTA)
  // Probing points may be verified at compile time within the radius
  // using static_assert(HYPOT2(X2-X1,Y2-Y1)<=sq(DELTA_PRINTABLE_RADIUS),"bad probe point!")
  // so that may be added to SanityCheck.h in the future.
  #define _MIN_PROBE_X (X_CENTER - DELTA_PRINTABLE_RADIUS)
  #define _MIN_PROBE_Y (Y_CENTER - DELTA_PRINTABLE_RADIUS)
  #define _MAX_PROBE_X (X_CENTER + DELTA_PRINTABLE_RADIUS)
  #define _MAX_PROBE_Y (Y_CENTER + DELTA_PRINTABLE_RADIUS)
#elif IS_SCARA
  #define SCARA_PRINTABLE_RADIUS (SCARA_LINKAGE_1 + SCARA_LINKAGE_2)
  #define _MIN_PROBE_X (X_CENTER - (SCARA_PRINTABLE_RADIUS))
  #define _MIN_PROBE_Y (Y_CENTER - (SCARA_PRINTABLE_RADIUS))
  #define _MAX_PROBE_X (X_CENTER +  SCARA_PRINTABLE_RADIUS)
  #define _MAX_PROBE_Y (Y_CENTER +  SCARA_PRINTABLE_RADIUS)
#else
  // Boundaries for Cartesian probing based on bed limits
  #define _MIN_PROBE_X (max(X_MIN_BED, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
  #define _MIN_PROBE_Y (max(Y_MIN_BED, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
  #define _MAX_PROBE_X (min(X_MAX_BED, X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
  #define _MAX_PROBE_Y (min(Y_MAX_BED, Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
#endif

#ifndef MIN_PROBE_X
  #define MIN_PROBE_X _MIN_PROBE_X
#endif
#ifndef MIN_PROBE_Y
  #define MIN_PROBE_Y _MIN_PROBE_Y
#endif
#ifndef MAX_PROBE_X
  #define MAX_PROBE_X _MAX_PROBE_X
#endif
#ifndef MAX_PROBE_Y
  #define MAX_PROBE_Y _MAX_PROBE_Y
#endif
```

marcio-ao, 2017-11-27:
@thinkyhead: I like it :)

fiveangle, 2017-11-28:
Thanks for implementing my suggestion @thinkyhead 👍 :-D

thinkyhead, 2017-11-29:
Happy to help!

closed 2017-11-29 by thinkyhead