MarlinFirmware/Marlin #7626: Unable to move to negative coordinates in Marlin 1.1.5

marcio-ao, 2017-09-06:
- Description: X and Y blocked from going below 0
- Expected behaviour: When X_MIN_POS and Y_MIN_POS is negative, it should be possible to move to that location.
- Actual behaviour: Motion is stopped at 0, regardless of the value of X_MIN_POS and Y_MIN_POS
- Steps to reproduce:
  - Set X_MIN_POS or Y_MIN_POS to a negative value
  - Do a G28
  - Attempt to move to X_MIN_POS or Y_MIN_POS

Note, disabling software endstops using M211 is a possible workaround to this problem, but according to the [documentation](http://marlinfw.org/docs/gcode/M211.html), the software endstops should allow motion to [XYZ]_MIN_POS

tcm0116, 2017-09-07:
Can you post your configuration files?

marcio-ao, 2017-09-07:
Well, unfortunately, we have so many printer models and toolheads that we don't use flat config files. The standard Marlin config files are here with placeholder variables:

https://code.alephobjects.com/diffusion/MARLIN/browse/devel/Marlin/Conditionals.h
https://code.alephobjects.com/diffusion/MARLIN/browse/devel/Marlin/Configuration_adv.h

But the placeholder variables themselves are generated by conditional logic here:

https://code.alephobjects.com/diffusion/MARLIN/browse/devel/Marlin/Conditionals_LulzBot.h

I suppose I could generate a made up Configuration.h to try demonstrating the problem.

GMagician, 2017-09-07:
You issue is present also moving from LCD menu?
If affermative have you tryed to call M211 S1 and see soft endstop echoed values?

marcio-ao, 2017-09-07:
@GMagician: The problem is also present when moving from the menu.

"M211 S1" says:

 echo:Soft endstops: On   Min: X0.00 Y0.00 Z0.00  Max: X280.00 Y280.00 Z270.00

This is consistent with what I am seeing. I am unable to go below 0.

@tcm0116: I've written a tool to capture the settings in Configuration.h and Configuration_adv.h using the C proprocessor. I have attached those for your review.

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1286110/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1286111/Configuration_adv.txt)

GMagician, 2017-09-07:
It seems that 'soft_endstop_min' are initialized with X_MIN_BED, Y_MIN_BED, Z_MIN_POS.
I think that in this project there is a very big mix about what are 'physical machine limits' and what are 'working area limits'.
Theorically machine should move from soft limits (or endstops) manually and limit to 0,0,BED_SIZE_X,BED_SIDE_Y only for action related to probe and print with care of some offsets if bed is not fully available.
I think that some action should be taken to uniform these limits and give them a well defined action

marcio-ao, 2017-09-07:
@GMagician: That is what I figured. I suppose having the start GCODE turn software endstops on and off makes sense from a usability standpoint. But the way it is currently implemented does break backwards compatibility, at least until everyone modifies their start GCODE to disable soft endstops prior to going beyond the bed. If this change is intended, then perhaps the [documentation](http://marlinfw.org/docs/gcode/M211.html) needs to be updated to reflect the change.

thinkyhead, 2018-02-09:
Near as I can tell by scanning the code this was unintended behavior and has been patched. If not, it should be fixed to allow movement to `Z_MIN_POS` via G move commands, if not also via LCD menus.

marcio-ao, 2018-08-20:
Closing this ticket out since the bug no longer is present in 1.1.9

closed 2018-08-20 by marcio-ao