MarlinFirmware/Marlin #7627: FR.

NSchnitzler, 2017-09-06:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

teemuatlut, 2017-09-07:
You can use [M43](http://marlinfw.org/docs/gcode/M043.html) to read a pin value.

You can look into TMC2130 or TMC2208 if you want easier current management through a command.

NSchnitzler, 2017-09-07:
I Have 5 TMC2208, but couldn't find anything about them in the configuration_adv. I found some code for the TMC26X/TMC2130, but they have an other communication protocol.

teemuatlut, 2017-09-07:
I've got a standing [PR for TMC2208](https://github.com/MarlinFirmware/Marlin/pull/6345) but it's almost 5 months old and I'd actually advice against using it. You can try my [master development branch](https://github.com/teemuatlut/Marlin/tree/master). That's where the most up-to-date TMC support is and I occasionally rebase it whenever I'm doing firmware work.

NSchnitzler, 2017-09-07:
Ok i will give It a try. One more question, don't you need a single Serial wire for each TMC2208 stepper driver in order to config each single device?

teemuatlut, 2017-09-07:
Ideally yes, one serial port for each driver. But you can also wire them in parallel if you don't care about read capabilities and this way the drivers will have the same configuration. However the write only mode is a bit untested because that's not my personal configuration. There's software serial capabilities if you can't use hw serial.

NSchnitzler, 2017-09-07:
I had a quick look at the TMC2208 Arduino library and your master dev branch. For me, i would prefer a single software Serial interface for each driver, so you can tune chopper frequency, active current, hold current, micro stepping and so on individually. The Aux-2 connector would fit nicely for that idea (single extruder printer). In principle i have the capabilities to add such a feature, but the whole project structure is a little bit confusing for me, because all files are really large. I am not really sure, wether it's really necessary to read back the values, but it is useful to program each driver on it own (reading back is (for me) not a must have).

teemuatlut, 2017-09-07:
And you can do that without modifications to the firmware. Set the correct pins and solder the wires. 
[image]

detoxity, 2017-09-08:
Hello, how i can cofigurate this firmware for TRAMS (tmc5130).
I use #define HAVE_TMC2130. and define pins. but its not work.

teemuatlut, 2017-09-08:
You can't. Use my [TRAMS development branch](https://github.com/teemuatlut/Marlin/tree/trams-updated) of Marlin. You also need [a supporting library](https://github.com/teemuatlut/TMCStepper). The current status of TRAMS is largely untested and I need another printer to better start testing.

detoxity, 2017-09-08:
In this branch when i click home x (in repetier host) my axis z come down. and printer does not respond to movement commands.

teemuatlut, 2017-09-08:
It may be that the current online version isn't up to date what I have locally. I'll try to upload a more recent version along with my own configuration some time soon. But as said very much WIP.
I'd also ask that you make a new issue at my repo if you have further questions or probems.

detoxity, 2017-09-08:
Thank you, I'm looking forward to your new version.

NSchnitzler, 2017-09-08:
@teemuatlut thanks for the dev branch, i had a look at it and it should do what i need. I will give it a try at the weekend.

ModMike, 2018-04-05:
How do I wire the TMC2208s in parallel mode?

boelle, 2019-02-19:
@NSchnitzler you still need this or has the built in support for tmc2208 done the trick?

boelle, 2019-06-22:
wrong label

InsanityAutomation, 2019-06-22:
Id say the current state of UART and SPI comms to these drivers covers this use case and this can be closed. @NSchnitzler feel free to chime in if thats incorrect.

NSchnitzler, 2019-06-23:
For me is everything working fine now.

closed 2019-06-23 by NSchnitzler