MarlinFirmware/Marlin #7630: ./../../../avr/bin/ld.exe: region text overflowed by 19524 bytes     Pls help

Shrikant888, 2017-09-07:
Hi all..

I am using Sanguino with Full reprap LCD. It seems code is exceeding.

I am not using Z Probe and Auto bed levelling. But when I commenting them. Its showing below error

                           set_bed_level_equation_lsq(plane_equation_coefficients, false);

Ok these are specs:
Sannguino
Full LCD
4 Steppers
3 Endstops
SD Card
USB

Pls tell me which portion of code is using more space. How can I reduce it...

Shrikant888, 2017-09-07:
Using Marlin Firmware

Shrikant888, 2017-09-07:
Pls reply

tcm0116, 2017-09-07:
Please post your configuration files.

fiveangle, 2017-09-15:
You can see some feature sizes on a similar Teensy2.0++ based 128k board in my commented config files here:

https://github.com/fiveangle/Marlin/blob/progmem_usage_comments/Marlin/Configuration.h
https://github.com/fiveangle/Marlin/blob/progmem_usage_comments/Marlin/Configuration_adv.h

But the only way for you to know the size is to enable/disable features and see the result (which is not hard).

My recommendations are to enable NO_WORKSPACE_OFFSETS, DISABLE_M503, PREVENT_COLD_EXTRUSION, etc, but 20K you'll never get there I think.

Have you considered using Octoprint or some other host-based mechanism to print over USB ?  The SD feature pretty much consumes lots of available PROGMEM and leaves literally no room for anything if you also have LCD enabled.

Alternatively, you can buy a Mega based board (which as twice the PROGMEM available).

My config above is for Printrbot Simple Metal (also with 128k) and has Bilinear levelling, Graphics LCD, and Linear_Advance feature (without SD card support).

Good luck !

-=dave

closed 2017-11-14 by thinkyhead