MarlinFirmware/Marlin #7633: Flashing Marlin doesn't set default steps correctly

Louis-Cypher, 2017-09-07:
### Bug Report
- Description: Marlin doesn't set values for default steps
- Expected behaviour: Uploading the image should set values
- Actual behaviour: I changed Z steps to 2560 instead of 4000 as well as Extruder values but the values didn't change after a successfull compile and upload. Setting values using M92 works fine.
- Steps to reproduce:
  - Flash 1.1.5
  - Change value for Z steps
  - Check default step settings on LCD.

KevMags, 2017-09-07:
Either clear EEPROM before upload or send "M502" (Reset to default/firmware settings).

Louis-Cypher, 2017-09-07:
Am I getting you right that this is intended behavior? I never had to do that before.

KevMags, 2017-09-07:
This only happens when you reflash with the same version of Marlin.  Flashing a different version will overwrite EEPROM.

Louis-Cypher, 2017-09-07:
Ok,  thanks, got it. But is this intended or not? In case yes this is no bug and I'll close this issue

psavva, 2017-09-08:
Agree that this is by design.  Please close this incident if you're happy with the explanation

Louis-Cypher, 2017-09-08:
Thanks a lot for the clarification and the great work you are doing!

closed 2017-09-08 by Louis-Cypher