MarlinFirmware/Marlin #7634: Bed still not Level after both G33 and G29 on 1.1.5

iamthebest22, 2017-09-08:
Apologize for my other post, was stupid of me and realized, but another issue popped up.

------------------------------------------------------------
### Bug Report
- Description: Even after G33 P7 (with a deviation of between 0.021 and 0.035) and G29 Bilinear (9 points) probing succesfully 90% of the bed and all the numbers and ensured G29 is AFTER G28 or M420 S1 is activated, bed still not leveled not this: 
https://www.dropbox.com/s/run93ehb0c03cvm/2017-09-07%2018.40.38.jpg?dl=0
https://www.dropbox.com/s/99zrmnav8ri3vtb/2017-09-07%2018.40.58.jpg?dl=0
[image]

Back is squished and the front too high
- Expected behaviour: Bed should be leveled
- Actual behaviour: Still not leveled and uneven first layer
- Steps to reproduce:
  - Started with G33 P7, take the numbers from the lowest deviation iteration and put it into the Endstop Adjustments and Tower adjustments in the Arduino and re-flashed firmware then M502, M500 and M501 to double checked. 
  - After that did G29 to make sure it hit as much of the bed as possible without going off the bed. Checked the numbers and M500 to save and M420 S1 to activate.
- Inputted the following as starting code for Cura:
G28 
G29 
M500
M420 S1
and the usual for priming nozzle

Printer does the G29 fine again but once it starts the first layer test, the above picture happens.

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
Here is the ZIP of both files
[Marlin_1.1.5 Config h..zip](https://github.com/MarlinFirmware/Marlin/files/1286893/Marlin_1.1.5.Config.h.zip)

Please help, I've spent over 30+ hours trying to bed level it and it still isn't leveled.
------------------------------------------------------------

Roxy-3D, 2017-09-08:
Your configuration.h file says:
```cpp
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING
```
So...  anything you did with a G29 is not real.   Please turn on one of the Mesh Bed Leveling systems and try again.

iamthebest22, 2017-09-08:
Oh crap that's the wrong one, the Bilinear one was turned on I uploaded the wrong one. I can tell you AUTO_BED_LEVELING_BILINEAR was on. I had uncommented it to try something out, but the above pictures were done with Bilinear turned on. These were the numbers given:
0 -0.284 -0.291 -0.239 -0.263 -0.200 -0.351 -0.415
1 -0.241 -0.161 -0.104 -0.094 -0.129 -0.204 -0.438
2 -0.214 -0.157 -0.082 -0.030 -0.146 -0.224 -0.394
3 -0.375 -0.220 -0.087 -0.034 -0.097 -0.242 -0.442
4 -0.114 -0.248 -0.137 -0.122 -0.100 -0.165 -0.286
5 +0.374 +0.689 -0.006 -0.069 -0.035 +0.437 +0.183
6 +1.178 +0.601 +0.046 +0.026 +0.011 +0.362 +0.780

Roxy-3D, 2017-09-08:
The Bi-Linear expert is very very busy doing https://github.com/MarlinFirmware/Marlin/issues/7622.   

Both Bi-Linear and UBL use a bi-linear interpolation of the Z-height correction within a given a Mesh Cell.   If you bring up the UBL system, I can help you with any problem you have.

iamthebest22, 2017-09-08:
Ah no problem. Sorry to sound stupid, but how do you bring up the UBL system?

thinkyhead, 2017-09-08:
@iamthebest22 It looks like the point at 0,6 is way off from the others. But the points around it also seem to be raised, so perhaps that corner is just unusual.

In any case, you can check how bilinear leveling is behaving at any time by enabling the `DEBUG_LEVELING_FEATURE`. Then you can use `M111 S32` to turn on extra debugging for homing and leveling during those procedures.

Once this is active do the following:

- Do your bed leveling procedure, as usual. Save and examine the log output.
- Save the mesh to EEPROM with `M500`, if desired.
- Make sure leveling is enabled with `M420`. If not, enable with `M420 S1`.
- Move around the bed with `G1` or your host.
- Use `G29 Q` to see how much Z adjustment is applied to the current XY.

As you move around the bed, observe the nozzle distance to the bed. If there are points that seem too close or far, use `G29 Q` to see what Z adjustment is being made and check against the Mesh Output to see if it makes sense.

The bilinear interpolation should work properly, given a good set of mesh Z values. But if you find any strange numbers, please let us know!

iamthebest22, 2017-09-09:
hi all, thanks thinkyhead, I'm doing UBL right now, and it feels either the bed leveling is being saved even though it says it is. 
So here is something interesting, i turned on M48 to make sure my BLtouch isn't bad, and it doesn't look like it. here are the results:
17:33:52.584 : N23 M48 X90 Y-25 P5*73
17:33:52.588 : M48 Z-Probe Repeatability Test
17:34:15.967 : Finished!
17:34:15.968 : Mean: -0.072867 Min: -0.085 Max: -0.060 Range: 0.025
17:34:15.968 : Standard Deviation: 0.007904
17:34:15.969 : X:90.00 Y:10.00 Z:52.40 E:0.00 Count X:16524 Y:24547 Z:21743
17:34:51.569 : N24 M48 X0 Y115 P5*104
17:34:51.573 : M48 Z-Probe Repeatability Test
17:34:51.575 : ? (X,Y) location outside of probeable radius.
17:34:58.148 : N25 M48 X0 Y110 P5*108
17:34:58.153 : M48 Z-Probe Repeatability Test
17:34:58.154 : ? (X,Y) location outside of probeable radius.
17:35:04.360 : N26 M48 X0 Y105 P5*107
17:35:04.365 : M48 Z-Probe Repeatability Test
17:35:04.366 : ? (X,Y) location outside of probeable radius.
17:35:12.303 : N27 M48 X0 Y100 P5*111
17:35:12.313 : M48 Z-Probe Repeatability Test
17:35:12.315 : ? (X,Y) location outside of probeable radius.
17:35:21.401 : N28 M48 X0 Y90 P5*88
17:35:21.405 : M48 Z-Probe Repeatability Test
17:35:21.407 : ? (X,Y) location outside of probeable radius.
17:35:31.992 : N29 G1 X0 Y110*18
17:35:49.316 : N30 M48 X-80 Y-45 P5*97
17:35:49.320 : M48 Z-Probe Repeatability Test
17:36:12.879 : Finished!
17:36:12.883 : Mean: 0.051004 Min: 0.051 Max: 0.051 Range: 0.000
17:36:12.883 : Standard Deviation: 0.000000
17:36:12.885 : X:-80.00 Y:-10.00 Z:52.40 E:0.00 Count X:24846 Y:18101 Z:21077

Now this is interesting. As you can see X0 Y110 can't be probed even though it should be able to on my Delta Printer, Tried G29 Q to see but it gave me this in Repetier:
17:38:38.196 : N31 G29 Q*81
17:38:38.210 : Invalid test_pattern value. (-1 to 2)
This is a pic of it at X0 Y110, also dropped it to Y90 and same thing
[image]

As for the UBL, here is a printout of it:
17:12:58.246 : N23 G29 A*66
17:12:58.260 : Unified Bed Leveling System v1.01 active.
17:13:09.967 : N24 G29T0*64
17:13:09.981 : Bed Topography Report:
17:13:10.005 : (0,10)                                                                           (10,10)
17:13:10.028 : (-114,114)                                                                        (114,114)
17:13:10.226 : .          .          .          .          .          .          .          .          .
17:13:10.226 : 
17:13:10.397 : .          .          .          .          .          .          .          .          .          .          .
17:13:10.397 : 
17:13:10.568 : .          .          .       1.694    0.402    0.219    0.327    0.720       .          .          .
17:13:10.568 : 
17:13:10.738 : .          .       0.524    0.053    0.032    0.099    0.094    0.190    0.334       .          .
17:13:10.738 : 
17:13:10.909 : .       0.120    0.001    0.010    0.012    0.010    0.062    0.080    0.062    0.108       .
17:13:10.909 : 
17:13:11.079 : .      -0.003   -0.002    0.034    0.035    0.013    0.035    0.010   -0.036   -0.036       .
17:13:11.080 : 
17:13:11.277 : 0.081    0.025    0.034    0.036    0.075    0.007   -0.015   -0.015  [-0.015]      .
17:13:11.277 : 
17:13:11.447 : .       0.140    0.109    0.061    0.082    0.095    0.059    0.035    0.058   -0.010       .
17:13:11.447 : 
17:13:11.618 : .       0.179    0.119    0.075    0.077    0.061    0.073    0.086    0.067   -0.007       .
17:13:11.618 : 
17:13:11.789 : .          .       0.100    0.054    0.060    0.045    0.168    0.115    0.067       .          .
17:13:11.789 : 
17:13:11.959 : .          .          .          .          .      -0.018       .          .          .          .          .
17:13:11.983 : (-114,-114)                                                                            (114,-114)
17:13:12.007 : (0,0)                                                                             (10,0)
17:13:54.423 : N25 G29 A*68
17:13:54.437 : Unified Bed Leveling System v1.01 active.
17:14:03.609 : N26 M420 S0*82
17:14:03.613 : echo:Bed Leveling Off
17:14:09.514 : N27 G29 A*70
17:14:09.528 : Unified Bed Leveling System v1.01 active.
17:14:17.325 : N28 G26 B52 H196 F1.75 L0.2 S0.4*118
17:14:17.340 : G26 command started. Waiting for heater(s).

As you can see UBL is active, but the print still does this:
[image]
[image]
[image]

I'm lost as to what's going on. Here is my newest config.h and config_adv.h
[Marlin-1.1.5.zip](https://github.com/MarlinFirmware/Marlin/files/1289427/Marlin-1.1.5.zip)

thinkyhead, 2017-09-09:
I'm not too familiar with UBL and Delta interaction, but it probably has a similar feature to the `G29 Q` option that I added for `BILINEAR`, so you may be able to examine how much leveling compensation is being applied at various points. I'll poke around, and perhaps @Roxy-3D can give some recommendations on how to troubleshoot UBL more deeply.

The other thing that's important is, of course, to make sure your delta is well-calibrated even before bed leveling, so that it isn't making any bowl-shaped movements. If your movement without leveling is already nice and flat, then you're set there.

As for why it's not probing some point near the edge, I know that in the `BILINEAR` code I had to take care to use code like `if (x <= edge + 0.0001)` because `float` numbers can round up or down by small amounts. If the UBL code isn't accounting for `float` inaccuracies in the edge comparisons, then this may explain why your edge point isn't getting probed.

iamthebest22, 2017-09-09:
wait, the only calibration i did before was using G33 P7 and using the numbers it gave me and putting it in the firmware, reflashed and then M502, M500 and m501 to check. The standard deviation was 0.021, the only thing i had to check was the diagonal rod length, which the maker provided.

iamthebest22, 2017-09-09:
@thinkyhead you're right, I only did G33 and didn't check the movement, right now even after G33 settings, the nozzle is sliding closer and hitting the bed on one side and moving up higher up higher on the other side. I thought G33 was suppose to fix this?

iamthebest22, 2017-09-09:
yeah after all that above, movement is sliding lower on one side and higher on the either on both X or Y.

iamthebest22, 2017-09-09:
Just an update, I test my Bltouch using M48, deviation was -0.007 first time and literally 0.000 2nd time, so it's not the probe. Right now I'm just confused why after all the calibration via G33 and checking the build (which seems to be fine) why it still hits the nozzle on one side while hanging up higher on the other side...

Here's an updated config.h zip 
[Marlin-1.1.5.zip](https://github.com/MarlinFirmware/Marlin/files/1289963/Marlin-1.1.5.zip)

iamthebest22, 2017-09-10:
Okay so I went back to Marlin 1.1.4, and the results were a little better after G33 P7. The odd thing i however, after G33 P7 and inputed it into the firmware, flashed, M502 and then M500, the nozzle is still crashing towards the x tower, and higher between Y and Z, but not as bad as before. The odd thing that i said is that even after G29 and printing, it looks like it's not being applied when I did a test print, which is super weird... here is my 1.1.4 config and my start code from Cura:
Cura start code: 
G21        ;metric values
G90        ;absolute positioning
M107       ;start with the fan off
G1 F200 E-2
G28   ;move to endstops
G29
G92 E0                  ;zero the extruded length
G1 F200 E6              ;extrude 5mm of feed stock
G92 E0                  ;zero the extruded length again
That doesn't look like it should turn off the Bilinear thing
Config: 
[Marlin-1.1.4.zip](https://github.com/MarlinFirmware/Marlin/files/1290184/Marlin-1.1.4.zip)

Here are the results after the Bilinear probing in 1.1.4, the weird numbers I was getting like the 1.5 something at one point on the bed is gone.
18:40:05.918 : Bilinear Leveling Grid:
18:40:05.918 : 0      1      2      3      4      5      6      7      8
18:40:05.921 : 0 +0.186 +0.172 +0.174 +0.196 +0.135 +0.137 +0.087 +0.047 +0.023
18:40:05.925 : 1 +0.201 +0.165 +0.173 +0.147 +0.139 +0.102 +0.048 +0.014 -0.013
18:40:05.925 : 2 +0.146 +0.128 +0.125 +0.097 +0.074 +0.047 -0.022 -0.064 -0.074
18:40:05.932 : 3 +0.099 +0.070 +0.076 +0.102 +0.063 +0.014 -0.024 -0.061 -0.101
18:40:05.932 : 4 +0.080 +0.076 +0.076 +0.070 +0.047 +0.017 -0.013 -0.038 -0.086
18:40:05.938 : 5 +0.087 +0.086 +0.064 +0.027 +0.037 +0.027 +0.022 +0.001 -0.033
18:40:05.938 : 6 +0.062 +0.068 +0.048 +0.013 +0.015 +0.023 +0.037 +0.040 -0.017
18:40:05.942 : 7 +0.021 +0.022 +0.023 +0.053 +0.039 +0.076 +0.082 +0.071 +0.053
18:40:05.942 : 8 +0.203 +0.070 +0.054 +0.098 +0.102 +0.101 +0.096 +0.115 +0.152
18:40:05.944 : X:49.00 Y:84.00 Z:52.25 E:0.00 Count X:15887 Y:20980 Z:24783

f4bul1z3r, 2017-09-14:
I have the same printer and have had the same issue with 1.1.0 the marlin version on the anycubic google drive. I had this exact problem after using BLtouch to level. But then I tried a newer version with a ported config file and now I'm having an issue with the probe endstop being triggered all the time

thinkyhead, 2017-09-15:
> That doesn't look like it should turn off the Bilinear thing

`G28` is specified to disable leveling. After `G29` procedure, leveling will be enabled. To be doubly-sure that it is enabled, add `M420 S1` to your start G-code. This can replace `G29` after you use `M500` to save your (bilinear) mesh.

> Here are the results after the Bilinear probing in 1.1.4, the weird numbers I was getting like the 1.1.5 something at one point on the bed is gone.

Very interesting. Not much changed between 1.1.4 and 1.1.5 in the core leveling code. If you have a chance to provide a couple more samples of the Bilinear Leveling Grid output from both 1.1.4 and 1.1.5 for comparison, I will look into the issue more closely as soon as time allows. (At the moment I'm very busy reorganizing the code for the Marlin 2.0 branch.)

If you want to get very down and detailed to get your mesh just right in the meantime, look at the `G26` procedure included with Unified Bed Leveling. This allows for more fine-tuning. Whereas with ABL you can only set individual mesh points by typing in the new value, the `G26` procedure allows you to edit interactively.

thinkyhead, 2017-09-15:
@f4bul1z3r If a switch is triggered all the time, check both the INVERTING and the PULLUP settings related to the switch in your configuration. Assuming connections are good, pins are set correctly, and the polarity is right, then the probe should operate.

thinkyhead, 2017-10-16:
For some probes it's important to make sure all PWM is turned off during the probe, otherwise you may get strange results. Try enabling the `PROBING_HEATERS_OFF` and `PROBING_FANS_OFF` options and see if the probe results are any better.

ZhuDaHai, 2018-02-14:
I am running Marlin 1.1.8 with Bilinear Bed Level and having the exact same issue.
Any updates since Oct?

thinkyhead, 2018-02-23:
@ZhuDaHai — Just a few. Generally these things come down to an off-angle probe.

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

closed 2018-02-23 by thinkyhead