MarlinFirmware/Marlin #7635: Inexplicable thermal runaway issue on print bed while cooling.

marcio-ao, 2017-09-08:
I have attached a log that shows a thermal runaway on the print bed. I am trying to understand why the error is occurring. What seems odd to me is that the temperature reaches the target temperature of 110 degrees at 07:52:22, but then it begins to drop and until a thermal runaway happens. I am trying  to understand why the temperature begun to drop, and why this would cause a thermal runaway (it would seem as if a temperature runaway would only happen when heating).

[TempRunawayIssue.txt](https://github.com/MarlinFirmware/Marlin/files/1288288/TempRunawayIssue.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1288284/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1288285/Configuration_adv.txt)

psavva, 2017-09-08:
What sort of heated bed are you using?
Running an SSR?

I see you're using PID for the heated bed, maybe try calibrate the bed PID or try Bang Bang

marcio-ao, 2017-09-11:
It looks like the runaway was caused because I had MAX_BED_POWER 206 and the values of WATCH_BED_TEMP_PERIOD and WATCH_BED_TEMP_INCREASE had been tuned for MAX_BED_POWER of 255.

I'll close this issue.

closed 2017-09-11 by marcio-ao