MarlinFirmware/Marlin #7639: X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER not working

vangalvin, 2017-09-09:
### Bug Report
- Description: ---
I have set up my Z Probe and it now has an offset to the extruder of -7 in the X and 30 in the Y and is 5mm below the extruder.

I set the following in the Configuration.h 
#define X_PROBE_OFFSET_FROM_EXTRUDER -7     // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 30     // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -5  // Z offset: -below +above  [the nozzle]

- Expected behaviour: ---
This should cause the probe to align to the center of the bed when probing. 

- Actual behaviour: ---
The previous version was working OK however the latest version these settings seem to be ignored and there is no offset in the X or Y however the offset for the Z is working.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1289434/Configuration.zip)

Zezoh, 2017-09-10:
I have the same issue , could someone help please.

fiveangle, 2017-09-10:
```
#define X_PROBE_OFFSET_FROM_EXTRUDER 0 //-7     // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 0 //30     // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0 //0.10  // Z offset: -below +above  [the nozzle]
```

vangalvin, 2017-09-10:
@fiveangle Sorry, I commented the changes out and reset them to 0 while I was designing a new sensor only for testing purposes. The 0's have been removed and the values are back but still no offset.

Zezoh, 2017-09-13:
No one wanna help 😑

benlye, 2017-09-13:
Un-comment this line:
```
//#define Z_SAFE_HOMING
```

LVD-AC, 2017-09-13:
Probe should align taking into account the offsets, however with calibration (like G33) the effector-nozzle should align (no offsets).

vangalvin, 2017-09-13:
@LVD-AC Strange, the earlier version G33 used the offset in the X/Y.

LVD-AC, 2017-09-14:
Earlier version was wrong with eccentric probes; that is called a bug, happens sometimes...

vangalvin, 2017-09-14:
I am just about to reflash and test it to make sure that I can offset the sensor to the center of the bed. fingers crossed!!!

vangalvin, 2017-09-14:
@LVD-AC No, still seems to be ignoring the probe offset settings.

LVD-AC, 2017-09-14:
G33 does not use offsets, G29, etc. does... so what commands are you using?

vangalvin, 2017-09-14:
I was using the inbuilt LCD command on the printer for the leveling the Delta however checking on it it appears that its G33 and not G29, When i use a G29 i get no response so I must have missed something when setting it up.

closed 2018-02-09 by thinkyhead