MarlinFirmware/Marlin #7641: [FR] Use marlin without shields

hunterd, 2017-09-09:
I'm very new in robotic and I connected my drivers directly to my Arduino Mega. I can make steppers move using stepper library but I would like to know if there is a way to use Marlin firmware without shields. I ordered Ramps1.4 but it will take 2 months to arrive.

Roxy-3D, 2017-09-09:
There are a lot of boards that have everything on them.   No shield.  Just a single board.

Did you check on eBay for an Atmega-2560 and RAMPS v1.4 board?   Somebody on eBay can get stuff to you within a week.

hunterd, 2017-09-10:
I already own an Atmega-2560 but I have not received RAMPS v1.4 yet. I ordered them yesterday on a chinese website (banggood) since I try to get the lowest price to make my Scara "from junks". The problem is that I live on Reunion island and shipments take from 1 to 2 month to come here from this website. I would like to try to make my Scara write things and I wanted to know if it was possible without a shield.

Roxy-3D, 2017-09-10:
If you have an Atmega-2560, it can't do anything without a 'shield'.    And in fact, the RAMPS shield that most people use also needs 'shields' for the stepper motors.
 
If you didn't do it already...  You need a step-stick for each stepper motor to plug into the RAMPS board.

p3p, 2017-09-10:
Well to be fair you don't "need" to buy a shield, but you do need to know what you are doing, have the appropriate filtering on the endstop, adc signals, route the correct pins to your stepper drivers .. use the correct MOSFETS to supply the heater elements, fans .. building the control board yourself you would spent more but end up (if you do your research) with a much better solution than the usual RAMPS board (they are not built well in general, but "just" well enough).

So to sum it up the cheapest solution will be to use a RAMPS sheild unless you have a stockpile of the components and the skill to build the needed circuitry standalone, but any questions would need to be directed at an electronics forum not here I'm afraid.

hunterd, 2017-09-11:
Alright. Thank you for your help. I have l298n drivers so I guess it will do the job since it's already doing what I want whitout a RAMPS shield.
So I guess there is no easy way to use Marlin firmware without a shield since we have to fill shield pins instead of using stepper library directly like this :
`X1 = stepper(STEPS, 8, 9, 10, 11);`
Wouldn't it be interesting to ask this feature request?

hunterd, 2017-09-11:
Maybe there would be functions to edit so it would send the four signal in order to make my steppers turn instead of sending the direction and the steps?

RalphSch, 2017-09-25:
Marlin doesn't know "shields", it only knows which pin each signal has to use ;) Because this assignment is the (main) difference between all the controllers and shields.
The arduino stepper library is the simplest way of directly controlling one stepper without coordinating multiple steppers and things like acceleration and jerk. Thats why people wrote firmware like grbl to control cnc machines and marlin to use 3d printer. And they all use gcode because that's the easiest way of controlling multiple steppers at once.
If you just want to draw something, than attach the X, Y, Z steppers and the X/Y/Z endstops to your Arduino (look in pins_Ramps.h for pin assignment), take a picture and convert it to gcode, the send this to your scara .... and it will print it ;)

closed 2018-02-09 by thinkyhead