MarlinFirmware/Marlin #7644: Tevo Tornado not making Marlin Firmware open source

kimhee1990, 2017-09-10:
The Tevo Tornado is using a Marlin Firmware however they are not making it open source. That is considered violating the open source law of Marlin. Any solutions? [](https://m.facebook.com/groups/241512023021682?view=permalink&id=273401006499450)

Roxy-3D, 2017-09-10:
The first step is make a formal request that they provide the current (shipping) source to their customers.

kimhee1990, 2017-09-10:
Someone has already done so but the CEO declined

Jinxycob, 2017-09-11:
maybe a few emails from the marlin Dev team could clear this up, its striking a startup like Tevo would do this. "kimhee you have the email addr. spare  a willing Dev from having to search for it"
  did the printer come with an Sd card ? is it on there.

kimhee1990, 2017-09-11:
It did come with an SD card however it's not in there

marcio-ao, 2017-09-11:
The following may be a resource: http://gpl-violations.org

dot-bob, 2017-09-12:
They have released the firmware.  Right now it is posted in there facebook group under the files section.

Roxy-3D, 2017-09-12:
> it is posted in their facebook group under the files section

Can you provide a link?   I was trying to check it out and I can't find the 'Files' section.

dot-bob, 2017-09-13:
The firmware can be downloaded here but you will need to join the TEVO Tornado Owners group: https://www.facebook.com/groups/TEVO.Tornado.Owners/274134936426057/

itskeithb, 2017-10-27:
Here's a non-Facebook link to the Tevo Tornado firmware file that @dot-bob posted above: https://file.io/fTfKNc

bogaskilop, 2017-12-30:
This. Fb group is not egist any more ? (It has a lot of users, now it shows me only 5 and i'm not in that group lol but why? I was in that fb group). Non fb link is also down :(

jannis77, 2018-01-26:
http://tevo.cn/software-download.php