MarlinFirmware/Marlin #7646: PT100 and Re-Arm

apballard, 2017-09-10:
Hi All

I'm just trying to find out if anyone has managed to get pt100 working with re-arm?

I'm struggling so just trying to find out if its me or something else.

I need to do more tests, but I'm getting temperatures that are about 20 degrees too high to what the voltage output is actually giving.

There should technically be no issues, the temperature curve in the thermistor 20 file should be fine and should not change from what I can see.  I'm getting the right voltage, albeit lower because the ref voltage is 3.3v.

Anyway, just trying to check if its me or if someone else has a pt100 they can test with the re-arm and let me know if they get it working.

Bob-the-Kuhn, 2017-09-10:
I haven't tried it on Re-ARM but from my 2560 experience ...

Sounds like you're running a PT100 amplifier into a standard analog input.  The PT100 amplifier is pretty wimpy and can't properly drive the pull up resistor on the standard analog inputs.  

The usual solution is to use one of the analog inputs that doesn't have a pullup.

On the Re-ARM the analog inputs are very noisy.  You'll need to put a filter on it.  See issue #7585 for some software filter options.

apballard, 2017-09-11:
@Bob-the-Kuhn thanks so much! Yes I think that's the problem. Great suggestions. So do you recommend a cap on the input?

I moved originally from aux1 a3 to the satandard analog input. I will try that again, since I did fiddle around a bit with the pins in pinout. In that case I remember getting 100 degrees. But I'll fiddle and let u know how it goes.  Do you know off hand if the a3 input will do the trick? I'll dig up the data sheet of lpc chip. On the 2560 that would be the normal input for a pt100.

Bob-the-Kuhn, 2017-09-11:
Re-ARM has done violence to the RAMPS 1.4 pinout on the connectors.  See the [RepRap Wiki](http://reprap.org/wiki/Re-ARM) for details on the pinout.  There just weren't as many I/O pins on the new chip so the designers had to be creative.  Some pins are now no-connects.  Some pins are shared with other connectors.

If you're not comfortable installing the software filter then a 100uF cap is the next best option.  The three standard analog inputs have 10uF caps on them and they have serious noise problems.

p3p, 2017-09-11:
There is also an ADC hardware bug that causes very erroneous reading when toggling a pin directly next to the ADC pin you are reading, and just generally the ADCs hate stepper and PWM noise, so the software filter will be required for a clean signal. I will be adding it as default once we have a clear place to target 2.0 PRs again.

kfazz, 2017-09-11:
The ReArm boards are poorly designed and very fragile. I just fried my second one. The bed thermistor wire broke at the bed and I soldered it back on. Plugged the printer back in to get some magic smoke and all my ADC channels are fried except one. They read ~300 degrees with nothing plugged in. I've never had such bad luck with any board before. I'm planning on using the remaining ADC channel for the bed, and adding an SPI thermocouple for the hotend.

Bob-the-Kuhn, 2017-09-11:
Fried the analog channels?!!  Ouch!

Double check your wiring.  When setup as an analog input the pin can only handle 3.3V.  When setup as a general purpose I/O it is 5V tolerant.

apballard, 2017-09-11:
> Re-ARM has done violence to the RAMPS 1.4 pinout on the connector

 @Bob-the-Kuhn  Almost died laughing

I'm gonna look into it tomorrow, will let you know what I find.  I'll use the software filter and test.

As it is I recall I was getting pretty stable reading, just an unexpected one, not one congruent to the actual voltage measured.

I have few questions, what resolution is the adc configured for? Can it be set higher if needed, although notnsure that would make a difference. Another is can the internal pull-ups be disabled or are they disabled as soon as the pin is made an analog input? Could not work that out from datasheet.

apballard, 2017-09-11:
Wondering if the mks sbase board from china is any good. Can't be worse? Can it? Have u had experience with it?

Bob-the-Kuhn, 2017-09-11:
The ADC is setup for 10 bits but I expect you won't get that level of performance out of it.

The digital I/O logic is disconnected from the pad when configured as an analog input so the pullups within the chip have no effect.

I don't remember anyone mentioning the mks sbase board in the 32 bit threads.

I also don't remember complaints about the quality, just about the lack of support documentation.  You may end up mapping out the board using the M43 commands.

apballard, 2017-09-12:
Well let me see tonight what my tests reveal.

maybe as an alternative I'll use an i2c adc like this? Could potentially put both heated bed and extruders on it.

[http://www.ebay.com/itm/16-Bit-I2C-ADS1115-Module-ADC-4-Channel-With-Pro-Gain-Amplifier-For-Arduino-RPi-/263087926610?hash=item3d41432d52:g:YooAAOSwKoRZaF8Q](url)

apballard, 2017-09-12:
Ok, so here are my findings. First off I'm not terribly good at electronics, but I get by.

By connecting the pt100 to the standard t0 of the ramps board, think that's linked t p0.23 on the lpc chip, adc0, I get a stable reading, but always 30 degrees too high. This stumped me. So instead of 30 degrees I'd get a steady 59/60.  What made it worse is by fiddling around, plugging and unplugging things, at one point I had a stable 49. No software change.  After rearranging the cables I find that it returned back to 59/60.

I also tested the voltage coming from the amplifier and found that the voltage although a little higher than what it should be nothing  to what it  would be reading for that temperature. So I then tested the person with the amplifier connected and found that there was an additional 50 mV.  I'm not sure where the additional 50 mV comes from, but it returns back to the expected voltage when I unplug it. Another thing that  strikes me as odd, is that when I unplug the sensor, the pin reads 3.3v.

So I fished around for another adc pin, found one on aux 2 of ramps, maps to p0.26, adc3. Configured the pins and immediately I was getting readings in the 30 degree range, but this time it was bouncing around quite a bit, between 30 and 40. Software filter did not help. A 100uf cap from the sense pin to ground produced perfect stable results at 31 degrees, in line with my heated bed.

p3p, 2017-09-12:
The standard RAMPS temp connections are setup to read thermisters (variable resistance) so there is extra circuitry [(it's a voltage divider with a 4.7k pullup and the thermister)](http://rigidtalk.com/wiki/images/8/82/RAMPS1_4schematic.png) as well as the filter so you wouldn't get the correct readings with a variable voltage sensor. I'm glad you managed to get a good reading from the extra ADC pin (with the cap, 100uf is a big big but slow response isn't really a problem for temperature readings) the software filter will still be needed during a print to compensate for the noise more than likely.

Bob-the-Kuhn, 2017-09-12:
The 4.7k pullups on T0, T1 & T2 usually cause a reading that's high by 20C with a PT100 amplifier.  High by 30C is unusual but believable.

apballard, 2017-09-13:
@Bob-the-Kuhn @p3p  Thanks for your guidance and insight!

So I recommend using A9/D63 on the ramps board, set the pin to 3 in the pin file and put a cap on it for anyone else using re-arm with pt100.

This weekend I'll finalize my testing, see what effects the large cap has and then maybe reduce it from 100uf and start printing hopefully.

closed 2017-09-13 by apballard

kuhner, 2018-11-14:
@apballard How did this come out? I am looking into setting up a PT-100 on my re-arm. This thread is basically all I have to go on.

khomouda, 2019-02-10:
> 
> So I recommend using A9/D63 on the ramps board, set the pin to 3 in the pin file and put a cap on it for anyone else using re-arm with pt100.

Did you use it with 5v through the amplifier board?