MarlinFirmware/Marlin #7647: Encoder not working ( RepRapDiscount graphical contr clone JHD12864F )

BartDePauw, 2017-09-10:
Dear,
Seems this LCD board is not the easyest to setup.
Took me already a while before I had de screen working.
The push functionality of the encoder is working, as I can select the first menu ( where is starts on : 'Info screen' ) , and it goes back to the main screen as I push it again.
Beep is also working at push.

Do I need to change somewhere the port used for this encoder ( as was the case to have the screen working ) ?

Attach a ZIP of my `Configuration.h` as TXT file.

Kind regards.

PS : it would be great to have a 'encodertest.ino' that gives a beep when rotating.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1290923/Configuration.txt)

Bob-the-Kuhn, 2017-09-10:
Sounds like BTN_EN1 and/or BTN_EN2 are not setup correctly.  Look in the pins_YOUR_BOARD.h file and make sure they are set for your hardware configuration.

BartDePauw, 2017-09-13:
After some more analyzing and measuring I could find that there is about 1.7Mohm pullup to vcc ( should this be to 3.3 or 5 V for Arduino Mega ?) for the 2 pins used by the rotary encoder.
This is far to high I think, pullups are normally around 10kohm.
I found also that internal pullups can be activated on port 31 and 33 by using pinmode INPUT_PULLUP in place of INPUT. Where can I find these in Marlin ? It would be the easiest first check I can do before adding ( soldering ) two resistors to the rotary pins

closed 2017-11-14 by thinkyhead