MarlinFirmware/Marlin #7649: Parking Extruder Setup - Need help please

Scachi, 2017-09-11:
Hello,

I am trying to get a parking extruder setup working with Marlin 1.1.5 and I think I am missing some steps to configure and/or use it correctly. 

Short description:
I have configured the PARKING_EXTRUDER parts and assigned pins to SOL0 and SOL1.
I can activate the magnets manually with "M42 P15 S255" (left)  or "M42 P19 S255" (right)
What else do I need to configure to get it really working ? Or do I still have to setup a custom gcode to run on tool change and I am wrong thinking that this configuration shouldn't need any custom tool gcode ?

Any help is welcome.

Details:
I have an "Anycubic Prusa I3" that comes with a trigorilla board.
My magnets (5V Grove electro magnets) are connected to pins 15 (y-max) and 19 (z-max).

Extruder1 carriage (docks via magnet pin15) is on the left of the x-axis (min).
Extruder2 carriage (docks via magnet pin19) is on the right of the x-axis (max).
The magnet sits in between, driven by the x stepper motor.
Both extruder are adjusted to the same Z-offset.
```
x-axis:       |-E1-----Xcarriage------E2-|
positions:    0.......................210

```
The relevant Configuration.h lines:
```
#define PARKING_EXTRUDER
#if ENABLED(PARKING_EXTRUDER)
//  #define PARKING_EXTRUDER_SOLENOIDS_INVERT           // If enabled, the solenoid not magnetized with applied voltage
  #define PARKING_EXTRUDER_SOLENOIDS_PINS_ACTIVE HIGH  // LOW or HIGH pin signal energizes the coil
  #define PARKING_EXTRUDER_SOLENOIDS_DELAY 0        // Delay (ms) for magnetic field. No delay if 0 or not defined.
  #define PARKING_EXTRUDER_PARKING_X { 1, 210 }     // X positions for parking the extruders
  #define PARKING_EXTRUDER_GRAB_DISTANCE 1            // mm to move beyond the parking point to grab the extruder
  #define PARKING_EXTRUDER_SECURITY_RAISE 5           // Z-raise before parking
  #define HOTEND_OFFSET_Z { 0.0, 0.0 }                // Z-offsets of the two hotends. The first must be 0.
#endif
```

```
My relevant pins lines (edited pins_RAMPS.h):
//
// Limit Switches
//
#define X_MIN_PIN           3
#ifndef X_MAX_PIN
  #define X_MAX_PIN         2
#endif
#define Y_MIN_PIN          14
//#define Y_MAX_PIN          15 // now used as SOL0 - left extruder1
#define Z_MIN_PIN          18
//#define Z_MAX_PIN          19 // now used as SOL1 - right extruder2

#define SOL0_PIN 15 // parking hotend
#define SOL1_PIN 19 // parking hotend
```

Any idea what I am  missing ?

Scachi, 2017-09-11:
I started from scratch with the wiring for this setup and everything is working now.
There must have been a bad connection or something like that.

closed 2017-09-11 by Scachi

kasper30, 2018-11-25:
Hello
I would like to create the same system that you "dual parking extruder" would it be possible to have a copy of your configuration files (marlin file) that will serve as an example to create mine according to my system.

kasper30, 2018-11-25:
Thank you I will see what I can get from your files sanchant that I am under 1.4 ramps mounted on anet a8 modified aluminum chassis.
If I have configuration problems I could use your services?

Scachi, 2018-11-25:
I can't help you, I'm sorry. I have absolutely no experience with ramps.