MarlinFirmware/Marlin #7650: [FR] Embed the enabled plain text #defines from configuration files

dot-bob, 2017-09-12:
This is more of a new feature request / discussion.  I wonder if we should craft a way to embed the text of all the enabled #defines of the configuration.h and configuration_adv.h files in the firmware.
This would be similar to how you can retrieve the configuration of a compiled Linux kernel.

Perhaps we could craft it in a way that it would be beneficial for manufactures to leave it in or to make some essential functions rely on part of it.  For instance make the string table depend on portions of it.  This would help with the hit to the added space and make it beneficial to manufactures to leave it in.  Or we can go as far as if it isn't left in or at least partially then the firmware may miss functionality and or advanced features.

This might help combat the manufactures that don't supply the configuration or firmware.  But at the very least help you determine what settings and features are compiled in.

-Rob

oysteinkrog, 2017-09-12:
This is a good idea, but I think many of the 8-bit boards have very limited storage space, so that may be a challenge/constraint.

fiveangle, 2017-09-20:
Agree, cool idea: kind of like they embed cue sheets / song names for full-length MP3/FLAC files in order to fully restore to original configuration.

The only way I could envision this working is if we maintained a bitmap tied to each release so it wouldn't add hardly any resource usage, and even then once mfrs get ahold of it, that would go out the window and would be useless.

The full text strings of the compile-time config could probably be embedded for 256k boards without much issue, but since on 128k boards every bit is accounted for, even 2k for a "mandatory" feature that added no working value would be a hard sell.

thinkyhead, 2018-02-09:
It's a tall order. One more thing that would need to be maintained in perfect synchronization with all settings, and which would need to have a structure that can be extracted. Even as configuration options evolve and change across versions… Unless some homebound person is passionate about this, it's unlikely to be maintained very well.

boelle, 2019-06-22:
should this not have the feature request label?

sjasonsmith, 2020-11-10:
@rhapsodyv has posted one possible solution to this: #20081

thisiskeithb, 2022-01-21:
Added in https://github.com/MarlinFirmware/Marlin/pull/21321

closed 2022-01-21 by thisiskeithb