MarlinFirmware/Marlin #7652: Z Offset wont change when negative numbers are entered

eMotion2, 2017-09-12:
So I now implemented an induction probe as Zmin Endstop.
When I set either the **Z_CLEARANCE_DEPLOY_PROBE** or the **Z_CLEARANCE_BETWEEN_PROBES** Z0 is shifted down by that amount. However I want Z0 to stay where it is and just lift the probe during probing.

And then I have the issue that I cannot go beyond the Endstop Trigger Point with the Z offset. So negative Numbers are saved in the ROM but they aren't executed. It seems like there was a config called **ENDSTOP_ONLY_FOR_HOMING** which handled excactly this, but this now has changed to **ENDSTOPS_ALWAYS_ON_DEFAULT**. However this doesn't change anything wether I enable or disable it.

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1296387/Config.zip)

ArchGalileu, 2017-09-13:
Hello @eMotion2 

Comment out `#define MIN_SOFTWARE_ENDSTOPS`

and you can go "negative" ;)

Best regards
ArchGalileu

eMotion2, 2017-09-13:
@ArchGalileu 
Thanks for the answer <3
This helped me with this negative thing.
However this doesn't solve my first problem that when I home the z-axis the Probe then goes down by the amount I set in **Z_CLEARANCE_BETWEEN_PROBES**. So the ZPosition is shifted down.

empakoso, 2017-09-20:
Is your M851 reporting the correct distance between nozzle and point at which the z probe signal is true?

eMotion2, 2017-09-21:
Nope, it reports what ever I set the z offset to, but it doesn't change anything in the position.

empakoso, 2017-09-21:
I would be interested to see you configuration.h file.
Have you printed a calibration cube on your hardware at one point? Asking to rule out your steps/mm values.
Your probe offsets might be a little off.

If your probe offsets (X,Y) is set correctly the probe will go to the center of the bed and sense the bed position (not your nozzle).
If done correctly the Z axis will then see the Zaxis limit switch produce a TRUE value to your Z- input.  At that point the Zaxis stepper motor count is marked with whatever the value you report from M851.

Just curious, is your enstop approaching the bed from a high enough position? Sometimes the sensors need a distance to turn off (Hysterysis).  My inductive Sensor needed more than 5mm height reached before it would reset then be ready for another approach to the bed.
With that said let us know if your input is changing state from FALSE to TRUE as expected.

Does your Z offset value change immediately after adjusting your M851 Z(value) to a more positive value? If you execute an M851 Z(reported value+2.00) should step that value and raise the nozzle immediately.

Quick repeatable method I use to set my M851 value;
I recommend, heating bed and nozzle for more than 5min or things get all screwy because of thermal expansion.  After a G28, The nozzle should be moved to the bed center.  Place a piece of paper under the nozzle.  Command a G0 Z0.09 and the nozzle should just begin to grab the paper.  That’s when you know that M851 is correct.
Also Zaxis backlash could be a problem so I also recommend testing your M851 until it consistently grabs the paper by doing the following commands.  I use this set of commands to get it right after the geometry has changed between my nozzle and point where my sensor triggers (assuming a 200x200mm bed)
G27
G28
G1 Z5
G1 X100 Y100 F4000 ;have paper under nozzle
G1 Z0.09 ;check for grab (adjust M851)
G28 Z
; then Repeat back where I said G1 Z5
; do this several times until you get
; at least a consistent M851 value
; within a 0.15mm setting you are happy with
M500 ; save it power cycle and test again.

eMotion2, 2017-09-21:
My config can be seen in the first message.
So the overall functionality works, like moving directions, steps per mm and all of that works. The probe is working correct and triggers when it senses the bed.
My problem is that after homing the nozzle is in the middle of the bed, however the printer assumes it to be at home position (so Xmax, Ymax). 
I set an **PROBE_DEPLOY_DISTANCE** is 3mm so the sensor doesn't trigger right away. But after probing the nozzle moves 3 mm up again and the z0 is set at the wrong point.

empakoso, 2017-09-21:
You have an inductive Sensor attached to a servo? Just questioning why you have a deploy height. ( I assumed you have a stationary Z probe)

empakoso, 2017-09-21:
I have my Z minimum set to -2mm so that I can move beyond my probe position.  It really seems like your M851 value is not good.

eMotion2, 2017-09-21:
I have a stationary Z Probe, but in Marlin it says   `Inductive probes need space to keep from triggering early.` So I set both to 3.
`#define Z_CLEARANCE_DEPLOY_PROBE   3 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  3 // Z Clearance between probe points`.
I think somehow there is a combination of settings just working wrong in my code...I tried and changed every variable which had to do with z probing but nothing really changed anything.

empakoso, 2017-09-21:
Here are the settings that affect your probe
#define USE_ZMIN_PLUG
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define FIX_MOUNTED_PROBE
#define PROBING_FANS_OFF //Turn fans off when probing
#define DELAY_BEFORE_PROBING 200 //(ms) To prevent vibrations from triggering piezo sensors
#define X_PROBE_OFFSET_FROM_EXTRUDER 21 //you have to measure your probe to nozzle distance// X offset: -left +right [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 36 //You have to measure your probe to nozzle distance// Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.260 //M851 reports this value// Z offset: -below +above [the nozzle]
#define PROBE_DOUBLE_TOUCH 
define Z_CLEARANCE_DEPLOY_PROBE 1 //When finished probing with my Zoffset above the probe is left at 1.26 above the bed after g28
#define Z_CLEARANCE_BETWEEN_PROBES 5
Play with these values and you should be good with your probe

eMotion2, 2017-09-23:
Ok so first of all. Thank you really really much. I now set some values and it works for printing. When I controll it through repetier there are still these wierd things that I described above, but that is really no big deal for me. To sum it up: the z-Probe and the Offset works. 👍 
How do I now implement bed leveling to happen for the prints?

empakoso, 2017-09-23:
Great news!
Start with a simple ABL by enabling it in the firmware (remove the two “//“ marks in front in your configuration.h file) so the line looks like this:
#define AUTO_BED_LEVELING_BILINEAR
Next in your startup gcode:
G28 ; to Home your three axis’
G29 ; to find your ABL (BiLinear) mesh
M420 Z10 ; fade your mesh at 10mm height
M420 S1 ; enble the mesh
M117 ABL active!! ;; go ahead and print!

eMotion2, 2017-09-24:
Oh man I cannot thank you enough. This is easily the best improvement I made to my printer.  :)
And now that it works this issue is closed. 
Thanks

closed 2017-09-24 by eMotion2

empakoso, 2017-09-24:
Glad it worked out.