MarlinFirmware/Marlin #7654: Build with DOGLCD & MINIPANEL fails for Megatronics 2.0

jpjokela, 2017-09-13:
When I compile with `MINIPANEL` with `#define DOGLCD` and `#define MOTHERBOARD BOARD_MEGATRONICS_2` I get an error:
In file included from sketch/ultralcd.cpp:76:0:
ultralcd_impl_DOGM.h:189: error: 'DOGLCD_CS' was not declared in this scope
   U8GLIB_MINI12864_2X u8g(DOGLCD_CS, DOGLCD_A0); // 4 stripes
                           ^
ultralcd_impl_DOGM.h:189: error: 'DOGLCD_A0' was not declared in this scope
   U8GLIB_MINI12864_2X u8g(DOGLCD_CS, DOGLCD_A0); // 4 stripes
------------------------------------------------------------
Looks like this likely happens on stock build too.

The display in question is MKS MINI12864 V1.0.

I assume I'll have to use SW SPI? (already got Hello World onscreen by using separate graphics library example program). The relevant part of the code however seems to assume HW SPI
( ultralcd_impl_DOGM.h:  U8GLIB_MINI12864_2X u8g(DOGLCD_CS, DOGLCD_A0); // 4 stripes )

Maybe I'll get it working by modifying that line, but there should be more "clean" way to do this, I'd say :-)

jpjokela, 2017-09-13:
Maybe something like:
`#elif ENABLED(MINIPANEL)
  #if DOGLCD_MOSI != -1 && DOGLCD_SCK != -1
    U8GLIB_MINI12864_2X u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0); // 4 stripes
  #else
  // The MINIPanel display
  //U8GLIB_MINI12864 u8g(DOGLCD_CS, DOGLCD_A0);  // 8 stripes
  U8GLIB_MINI12864_2X u8g(DOGLCD_CS, DOGLCD_A0); // 4 stripes
#else`
to ultralcd_impl_DOGM.h?

Of course the defines should also be defined somewhere, and whatever should the default values be? No idea.

I added:
`#define DOGLCD_CS 31
#define DOGLCD_A0 32
#define DOGLCD_SCK 30
#define DOGLCD_MOSI 33`

to my Configuration.h and manually connected display to the relevant pins. I do have some adapter, but apparently it can't work with this particular display...

So now I successfully get image on my display, but haven't tried much else yet (got it working this far a few minutes ago, and the rotary encoder still isn't connected)

fiveangle, 2017-09-13:
You'll need to outfit your pins file to include the relevant mappings for your board when defining just MINIPANEL in Configuration.h (that define should take care of the graphics panel define so no need to explicitly define doglcd):

https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/pins_MEGATRONICS_2.h

Here is an example of what I added for the Printrboard_RevF to support MINIPANEL:

https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/pins_PRINTRBOARD_REVF.h#L171-L210

Once you get the pins file it all working as expected, submit a PR with your changes and you (and everyone else trying to use it wiht MEGATRONICS_2 will be set :)

thinkyhead, 2017-11-14:
Any issue with this now?

closed 2017-11-14 by thinkyhead

jpjokela, 2017-11-14:
Well, what was supposed to be just a simple (unrelated) "replace a few parts with alternative parts" operation, didn't turn out as simple as I thought it would...

I'll keep my current wiring until I can get the more critical issues sorted out, and will see about the adaptor & pin naming after that. No promises on any kind of schedule.