MarlinFirmware/Marlin #7660: Issues with Z Min Probe

f4bul1z3r, 2017-09-14:
**Description:**

When running M119 I can see that all endstops are functioning normally except the Z min probe.

**Expected Behaviour:**

I expect the Z min probe and the Z min endstop value to be The same.

**Actual Behaviour:**

Whenever I connect to the printer with repetier host, I get an error that the BLTouch has had an error. something like "stop was called blah blah blah reset with M999" that pops up in repetier

I believe this problem stems from the fact that M119 says that the Z min probe is always triggered.

It says that it is triggered when the leads are open and shorted out. 

What does change is the Z_Min value based on if the leads are shorted or not.

**Steps to reproduce:**

The issue appears even after a restart of the printer.

All the wiring is perfect since bed leveling worked (although not well due to another bug in the older version of marlin that I was originally using). Hoping that the issue was resolved in a later version, I port over my config and then realize that the printer will only error out whenever it is reset or whenever g29 is called.

GMagician, 2017-09-14:
The error appears when bltouch is not working properly (or disconnected), can't say more about you issue

f4bul1z3r, 2017-09-15:
This was a result of a configuration issue having unexpected consequences.

closed 2017-09-15 by f4bul1z3r