MarlinFirmware/Marlin #7661: [FR] How to Update/change firmware?

ProfessionalTrash, 2017-09-14:
(New to the whole github thing, so sorry in advance if anything is formatted incorrectly/not enough information provided.) I am using a Tronxy P802MA (running Marlin) controlled by a Raspberry Pi 3 running octoprint. The printer has a melzi2.0_v5 board. It has automatic be leveling enabled (does a g28 first to home all axes, then does a g29 that goes around to 9 points on the bed). The problem that I am having is that the automatic bed leveling does not really seem to help much. I know it is working because the z axis motors slightly move when the extruder is going side to side. I think it is currently set to planar bed leveling, which requires that the bed is flat, and compensates for tilt. My bed seems to be warped, so I need to enable automatic mesh bed leveling in order for the printer to correctly compensate. As I stated in the comments on this post: https://github.com/MarlinFirmware/Marlin/issues/5211#issuecomment-327952423 I cannot figure out what version of Marlin I need to update to, or how to do so. If someone could provide a complete step-by-step walkthrough of how to change the firmware, that would be very helpful.

Roxy-3D, 2017-09-14:
Move your changes to this branch's Configuration.h and Configuration_adv.h files...

And then follow one of the guides here to bring up the bed leveling system:  https://github.com/MarlinFirmware/MarlinDocumentation/tree/master/_features

ProfessionalTrash, 2017-09-14:
I'm not sure what you mean. I haven't made any changes to the existing firmware as I'm not sure exactly what it was and what needed to be changed to make it compatible with my printer. The few times I tried jumping into a firmware change I one main issue, which board compatibility, because I'm not sure what board I need to select in the Arduino IDE in order for it to connect and compile correctly.

Roxy-3D, 2017-09-14:
Tell your vendor you want the source code to the firmware that shipped on your printer.   You can then take those configuration.h and configuration_adv.h files and cross the settings over to the bugfix-1.1.x branch.    The configuration.h file you get from your vendor will have the mother board electronic controller specified in it.  You can tell Arduino to use that board when you build the firmware.

ProfessionalTrash, 2017-09-22:
Figured I should provide an update so this thread doesn't get closed due to inactivity; I sent the seller a message last week asking for the firmware package, got a reply a couple days later asking why I needed the firmware files. I replied explaining that I will be updating to the latest version of marlin in order to use features like mesh bed leveling, and that I need the original firmware version so I can copy over the critical settings. Sent that message Monday, no reply yet. I think its kind of weird that the seller is asking the reason for the reason I need the firmware and has still not replied. Is there any other way to do this without the original firmware?

Bob-the-Kuhn, 2017-09-22:
You should be able to get most of the configuration.h & configuration_adv.h settings by issuing a M503 command.

Be sure to save the original flash image before you download a new one.

ProfessionalTrash, 2017-09-22:
Ok so its just the stuff stored in the EEPROM that I need to copy over?

ProfessionalTrash, 2017-09-25:
So is there a different way I can figure out what board I need to tell arduino to use?

fiveangle, 2017-09-26:
By the way, because of the license Marlin is realeased under, the vendor is technically legally obligated to provide the source for the FW to you without question.

ProfessionalTrash, 2017-09-26:
Kind of what I was thinking. Do you have a link to where the specifics of the marlin license is?

fiveangle, 2017-09-26:
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/LICENSE

https://www.gnu.org/licenses/old-licenses/gpl-2.0-faq.en.html

ProfessionalTrash, 2017-09-26:
Thanks for that. Will get in contact with the seller again. Just a heads up question, I've looked around and some people apparently need to flash a bootloader and use a ICSP to flash the bootloader. Not sure what these are or why I need them, so if someone could explain that would be great. Also based on the discussion here: https://www.thingiverse.com/groups/tronxy/topic:10776  Irishobo made a comment (fourth from the bottom) that they got the original firmware for the P802MA with autolevel (same as mine is). For now I'll try that software and see how far I get.

ProfessionalTrash, 2017-09-26:
ok so the first issue I have is just trying to verify the firmware in arduino IDE gives this error in the SDBaseFile.h: exit status 1
using typedef-name 'fpos_t' after 'struct'
What needs to change in order for it to compile correctly?

fiveangle, 2017-09-26:
This is the information you should have been requested when you opened this, and it really is about the mimim info needed to make any useful comments (including the Config*):

https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/issue_template.md

But you're probably better off digging in with the community you're already connected to in order to get the original configuration ported by hand to the current Configs FW as Roxy suggested in the first response above.  They know your HW better than anyone here most likely.

ProfessionalTrash, 2017-09-27:
Unfortunately there doesn't seem to be much of a community for this printer. Thanks to everyone here for the help so far. I think most of the settings are set correctly between the old and new Configuration.h files (except for the changes I wanted to make). It seems I have to buy a usbasp in order to flash a bootloader to the board, then I'll try uploading the new firmware to the printer. So if all goes well, I shouldn't need to post here again. I'll check back once I get the usbasp and upload the new firmware. Thanks again to everyone for the help!

ProfessionalTrash, 2017-10-04:
Huge thanks for everyone here that helped point me in the right direction. I was able to use an USBISP to burn a bootloader to the board, and I can now upload the Marlin 1.1 firmware! 😄 I still have a few issues relating to bed heating, auto leveling, and the LCD screen. should I post the config files here or create a new issue?

Roxy-3D, 2017-10-04:
The bed heating will be the simplest to get going...    What's the problem?

ProfessionalTrash, 2017-10-04:
I have it set to use PID settings, and when i set it to heat to about 75c (i print PLA at 60 and ABS at 90 so im just using the average) it triggers the heating failsafe. I think this is due to how slowly it heats up (found another thread where someone had this issue and so they just disabled the failsafe to work around it, but i don't want to do that). Thing is, I had forgotten about having gcode set to send PID values upon printer connection, so the first time I connected it, it worked fine because I had a some pretty extreme values from trying to figure out if PID tuning would even work. I'll try a PID tune next, if it still does the failsafe i'll try disabling it until the PID tuning is done then retry heating it with the failsafe enabled.

emartinez167, 2017-10-04:
Try increasing the time it waits for the temperature to reach its value. For large beds like mine (RigidBot Big) that is what I had to do. It is in the configuration_adv.h file:

/**
* Thermal Protection parameters for the bed are just as above for hotends.
*/
#if ENABLED(THERMAL_PROTECTION_BED)
#define THERMAL_PROTECTION_BED_PERIOD 60 // Seconds
#define THERMAL_PROTECTION_BED_HYSTERESIS 2 // Degrees Celsius

/**
* Whenever an M140 or M190 increases the target temperature the firmware will wait for the
* WATCH_BED_TEMP_PERIOD to expire, and if the temperature hasn't increased by WATCH_BED_TEMP_INCREASE
* degrees, the machine is halted, requiring a hard reset. This test restarts with any M140/M190,
* but only if the current temperature is far enough below the target for a reliable test.
*
* If you get too many "Heating failed" errors, increase WATCH_BED_TEMP_PERIOD and/or decrease
* WATCH_BED_TEMP_INCREASE. (WATCH_BED_TEMP_INCREASE should not be below 2.)
*/
#define WATCH_BED_TEMP_PERIOD 180 // Seconds
#define WATCH_BED_TEMP_INCREASE 2 // Degrees Celsius
#endif

Regards,

Ernesto.

Roxy-3D, 2017-10-04:
Personally...  I think we have the Thermal Protection set too tightly.   I have to relax it on two of my printers too.    But then again...   I would rather some extra people have to edit their configuration data and consciously relax the Thermal Protection instead of burning somebody's house down.

ProfessionalTrash, 2017-10-04:
Alright thanks I will try that. What should I change it to? when im heating it all the way up to 90 it takes probably around 10-20 minutes (running a low budget, low power printer so thats why). I guess I should just increase it by about 60 seconds or so until it works?

Roxy-3D, 2017-10-04:
I'm not sure of the right thing to tell you.   With that long of a time period, it maybe you are better off turning the Thermal Protection off.   If the heaters are that unresponsive, it is hard to tell if they are working correctly.

fiveangle, 2017-10-04:
Turning protection off is never a good idea so I'm not sure where that's coming from.  If it takes 10 min to heat to 90 (which is about the same as my Printrbot before I upgraded from ATX to 13.9v PS) I just increased mine from 60 to 120 and it was fine.  If you're at 50 ambient and need to heat to 90, then that's 40 deg and default looks for 2 deg change in 60 seconds.  Changing to 120 looks for 2 deg change in 2 minutes, which should be absolutely fine for the slow 80-90 deg temp range.

emartinez167, 2017-10-04:
I doubled mine to 120 and it was fine.

Regards/Saludos,

Ernesto.

ProfessionalTrash, 2017-10-04:
Has been able to heat to 90 fine, it just takes a while. Not sure a UPS upgrade is what i'm looking for right now, since im not sure what exactly my board can handle (plus id probably add a mosfet). Probably an upgrade for the future, but right now i'm just focused on getting the firmware to work so i can actually print.

ProfessionalTrash, 2017-10-05:
So it seems I got the autolevel to work, but now I have an issue with the z homing. When I do a G28, it raises the z axis about 10mm or so, then homes x axis then y axis, then raises z axis 10mm again, then moves to the center of the bed (because of  z_safe_homing) homes the z axis, then after that it raises the z axis yet again, at which point the z axis position reads 0 and cannot be moved down any further because it is already at minimum. Changing z min to -20 makes it raise by about 30mm after homing at which point it reads 0 and cannot be moved down. Here are my files: 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1357874/Marlin.zip)

Roxy-3D, 2017-10-05:
> When I do a G28, it raises the z axis about 10mm or so, then homes x axis then y axis, then raises z axis 10mm again, then moves to the center of the bed (because of z_safe_homing) homes the z axis

Remember, there are multiple Z-CLEARANCE numbers.   One for safe movement, and one to deploy the probe.   You are probably seeing the various points where it need to switch modes.

```cpp
#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
``

ProfessionalTrash, 2017-10-05:
I have changed it to:
```#define Z_CLEARANCE_DEPLOY_PROBE   0 //10 // Z Clearance for Deploy/Stow```
```#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points```
This still raises the z axis before doing a G28 (which I am fine with), no longer raises it after homing x/y and before z (am fine with this), but still raises the z axis after homing z, setting z position to 0 after homing and AFTER raising, so the nozzle is about 10mm from the bed and unable to be moved down. Running G29 moves it to the first probe position, continues on through the probing procedure, then raises the z axis again after probing is done. By then z is set to around 5.4 (even though its the same position as when it raises after z homing), but can be lowered close to the bed (where it does read 0 at that point).

ProfessionalTrash, 2017-10-07:
Decided to open a new issue here: https://github.com/MarlinFirmware/Marlin/issues/7889

Roxy-3D, 2017-10-08:
>  and unable to be moved down.

Can you turn these off and see if the problem is still there?   I'm wondering if the software endstops are entering into this.
```cpp
// If enabled, axes won't move below MIN_POS in response to movement commands.
#define MIN_SOFTWARE_ENDSTOPS
// If enabled, axes won't move above MAX_POS in response to movement commands.
#define MAX_SOFTWARE_ENDSTOPS
```

ProfessionalTrash, 2017-10-08:
Still raises after homing z/autolevel probing, but can now be moved down afterwards. I believe the issue is related to the Z_CLEARANCE_BETWEEN_PROBES option, because it doesnt raise after homing z/autolevel probing. However this causes issues for autolevel probing as one might expect.

ProfessionalTrash, 2017-10-08:
So after some more experimenting, it seems that it does set the z position correctly (0 right at the bed), its just the moves afterwards that was confusing the host software that I was using (Repetier Host. Normally run off of octoprint, was just using repetier to test stuff out while it was connected to my computer). When it moves up after z homing, it sets z to 5mm if i send a M114. Does the same thing after z probing, so I set the Z_PROBE_END_SCRIPT to G1 Z0 so it returns to the bed. Will try out a print and see if it works.

closed 2017-11-15 by thinkyhead