MarlinFirmware/Marlin #7662: Trouble with Z offset output when calibrating level sensor

caioz1jp, 2017-09-15:
I am currently using and Anet A8 with the V1.0 board.

I recently upgraded from Skynet 2.3.2 to Marlins 1.1.5 and i am finding i cannot get a good z-offset calculation from the G30 command
------------------------------------------------------------
### Bug Report
- Description: When I run the following commands through Octoprint I appear to get an inaccurate Z offset value.
M851 Z0
M500
G28
G1 X110 Y110
-At this point I use Octoprint to manually reduce the Z height. I start with the standard 10mm down movement then use 0.1mm increments until the nozzle grabs the A4 paper.
G92 Z0
G30 X110 Y110
After the G30 I get an output that denotes where the Z position is when the probe triggers in relation to the zero point.
M851 ZX.Y - X.Y is the Z value from the output of the G30
M500

- Expected behavior: The expect behavior is the G30 z offset output should correlate to the distance I lowered the Z axis after the G1 X110 Y110 command. So if after the initial 10mm reduction, if I have to press the button 7 times at 0.1mm movement, set the new G92 Z0, G30 X110 Y110 should output -0.7--0.8

- Actual behavior: My last calibration required me to lower the Z axis by 7 clicks on 0.1 reduction, or 0.7mm. I set the Z0 with G92 then follow with the G30 X110 Y110 command. The output is always some where between 0.13-0.17.

If i use that value the nozzle is not at the proper height when printing starts. It is always too far away from the bed.

[config_h.zip](https://github.com/MarlinFirmware/Marlin/files/1304926/config_h.zip)

benlye, 2017-09-15:
Same as #7576 - the output of G30 changed in a recent release.  To go back to the old behavior you need to uncomment `#define NO_WORKSPACE_OFFSETS` in configuration_adv.h.

caioz1jp, 2017-09-15:
Thanks for the input. That solved the issue.

closed 2017-09-15 by caioz1jp