MarlinFirmware/Marlin #7663: Idea: G-code for contact digitizing

Gadzook8, 2017-09-15:
I wrote a python program that digitizes objects using G30 a bunch of times (fold down touch probe spliced into end stop circuit). I figure that if I added a G-code with the same functionality then it would be easy for anybody to do the same thing. Should I give this a try or is it too obscure for marlin?

fiveangle, 2017-09-15:
I'm not sure what value such a mechanism could add if incorporated directly into marlin, but as a host-side implementation, that's pretty cool.  I'm envisioning a sewing needle cup attachment for bltouch and hours upon hours of standard probing with probe debug output on would accomplish the same ?

Gadzook8, 2017-09-16:
I think the benefit would mostly be accessibility. For some users it's no big deal to make their own probe and use a host side solution, but a lot of people who need things digitized arent comfortable playing around with their machines. My vision was that if this functionality was included in marlin then manufacturers might start adding digitizing probes to their printers. Users would just select digitize on the lcd, enter a few parameters (boundaries, retract distance, resolution, etc.), and hit go. In a few minutes they come back, grab the sdcard, and that's it.

closed 2018-02-09 by thinkyhead