MarlinFirmware/Marlin #7664: [TMC2130] temperature prewarn gets triggered on reboot

dogtopus, 2017-09-15:
- Description: The temperature prewarn flag is set every time the printer reboots, even when the stepper driver is cold
- Expected behaviour: Prewarn flag is set only when the driver is hot
- Actual behaviour: Prewarn flag is set on every cold boot/reboot, even when the driver is cold
- Steps to reproduce:
  - Install the TMC2130 stepper driver and enable TMC2130 in Configuration_adv.h
  - reboot, use `M911` to check the flags

teemuatlut, 2017-09-15:
The TMC2130 otpw monitoring has been reworked since it was added to upstream Marlin. I'm mostly waiting on the v2.0 hassle to be over and then I can do another major TMC related update. Until then you can either ignore this or use my [development branch](https://github.com/teemuatlut/Marlin).

RalphSch, 2017-09-28:
Push :) Marlin 2.0 is out and I had to degrade my tmc2130 to hardwired configuration because I can't get the tmc code parts compiled on Due hardware. So it would be nice if this issue gets addressed and I will help if you need some testing and debugging (I have many 8bit controller and Radds on Due for 32bit).

p3p, 2017-09-28:
> Push :) Marlin 2.0 is out 
2.0 is in testing after a huge refactor, most definitely not out (as in released), there will be a 2.0.x branch when that happens and probably lots of tweets from thinkyhead ;)

RalphSch, 2017-09-28:
Okay, I mean bugfix-2.0.x is out and I think the biggest changes have taken place so fixing basic stuff (like tmc support) needs to be done to get stable code for release .... or to just get more people working and testing on the bugfix-2.0.x branch.

teemuatlut, 2017-09-28:
There isn't really any reason for them not to work with DUE as the SPI library should be compatible (I assume). What was the error message? Steps to reproduce?

vtuan10, 2017-12-20:
With Marlin 1.1.7 the printer shuts down on restart when `#define STOP_ON_ERROR` is enabled. Without it, the printer tries to decrease the current and even applies negative current settings, because it is trying to "cool down" the drivers.

My Debug output:

> Y driver error detected:
overtemperature
short to ground (coil A)
short to ground (coil B)
		X	Y
Enabled		false	false
Set current	800	800
RMS current	1436	795
MAX current	2025	1121
Run current	25/31	25/31
Hold current	25/31	25/31
CS actual		0/31	31/31
PWM scale	0	255
vsense		0=.325	1=.18
stealthChop	false	true
msteps		256	0
tstep		1048575	4294967295
pwm
threshold		0	0
[mm/s]		-	-
OT prewarn	false	true
OT prewarn has
been triggered	false	false
off time		0	15
blank time	16	54
hysterisis
-end		-3	12
-start		1	8
Stallguard thrs	10	10
DRVSTATUS	X	Y
stallguard			X
sg_result		0	1023
fsactive			X
stst		X	X
olb			X
ola			X
s2gb			X
s2ga			X
otpw			X
ot			X
Driver registers:
	X = 0x80:00:00:00
	Y = 0xFF:FF:FF:FF
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

teemuatlut, 2017-12-20:
You have a communication error on Y driver (register reads 0xFF:FF:FF:FF). Check your wiring.

vtuan10, 2017-12-21:
Thanks, was some weird wiring issue

thinkyhead, 2017-12-25:
@dogtopus, @teemuatlut Is this issue resolved yet?

teemuatlut, 2017-12-25:
Unfortunately it's not for me to confirm that. But I think it's safe to close this considering it was opened in September and we haven't heard from OP since.

dogtopus, 2017-12-25:
Actually this issue is probably invalid in the first place. I think I somehow failed to fully power-cycle the motor driver before I saw it raised prewarn at reboot.