MarlinFirmware/Marlin #7666: EXTRUDER_1_AUTO_FAN_PIN won't turn off

grabenm, 2017-09-15:
My EXTRUDER_1_AUTO_FAN_PIN turns on automatically when my printer is on instead of turning on when the set temp is reached.  However my EXTRUDER_0_AUTO_FAN_PIN functions correctly. Whenever my set temp is reached (100) I can hear the EXTRUDER_1_AUTO_FAN kick on a little higher, like its reaching its full speed (225).  Any ideas as to what could be causing the fan to stay on? 

**Configuration_adv.h**
#define EXTRUDER_0_AUTO_FAN_PIN 8
#define EXTRUDER_1_AUTO_FAN_PIN 6
#define EXTRUDER_2_AUTO_FAN_PIN -1
#define EXTRUDER_3_AUTO_FAN_PIN -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 100
#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed

-------------------------------------------------------------------------------------------------------------------

**pins.RAMBO.h** 
#define SDSS               53
#define LED_PIN            13
#define FAN_PIN            -1
#define FAN1_PIN           -1

#define FILWIDTH_PIN        3  // ANALOG NUMBERING

/**********************************************************
  Extruder_Auto_Fan Pins
  Extruder_Auto_Fan_0 8
  Extruder_Auto_Fan_1 6
***********************************************************/

VBen, 2017-09-17:
Can you post your complete config and the modified pins_rambo.h

grabenm, 2017-09-20:
@VBen Sorry for the delayed resposne.

[configuration.docx](https://github.com/MarlinFirmware/Marlin/files/1318883/configuration.docx)
[configuration.adv.docx](https://github.com/MarlinFirmware/Marlin/files/1318885/configuration.adv.docx)
[pins_RAMBO.docx](https://github.com/MarlinFirmware/Marlin/files/1318886/pins_RAMBO.docx)

I recently disabled my z_probe and ever since then I have been experiencing this issue.

thinkyhead, 2018-02-09:
Presuming solved.

closed 2018-02-09 by thinkyhead