MarlinFirmware/Marlin #7667: Unified bed leveling issue

f4bul1z3r, 2017-09-15:
When enabling bed leveling there is a compile error on a certain line. This is a bug report not of that compile issue, but after that line is commented out and 1.1.5 is uploaded to a delta 3d printer with bltouch and UBL enabled.

Expected Behavior:
I expect UBL to probe a grid on the build plate for automatic leveling and perfect layer adhesion.

Actual Behavior:
UBL when the G29 p1 command is issued, probes what appears to be random points in only about one quadrant of the circular bed. Most of the bed does not get probed and therefore the auto bed leveling doesn't work. Probing also does not follow the bilinear type of grid

Steps to reproduce:
send g29 p1 to the printer
observe weird behaviour

Roxy-3D, 2017-09-15:
Which code base did you get the compile error in?   Can you post your Configuration.h and Configuration_adv.h files and I'll try to reduplicate it.

But seriously...   We need to fix the compile error.   We can't just be commenting out code to make the error go away.

f4bul1z3r, 2017-09-15:
The error is in the g26 mesh bed levelling file. I am away from my computer at the moment but will provide more info when I can. But this issue is dedicated to the ubl not probing in a bilinear fashion not the compile error. I'll make a new issue for that.

fiveangle, 2017-09-15:
Did the issue template get jettisoned somewhere ?  It seemed to help out (slightly) by ensuring that submissions from reasonably savvy users at least contained the required bits.

@f4bul1z3r - as mentioned, it's near impossible to help you without seeing at least your Config*.h, the compile error, and exact line you commented out.

f4bul1z3r, 2017-09-15:
I don't expect any help for that issue in this thread. When I am able to get the config files I will start a new issue for the compilation bug.

fiveangle, 2017-09-15:
UBL does not probe in a standard grid pattern like other levelling mechanisms like BILINEAR, LINEAR, MBL, so that assumption is incorrect.  It should of course eventually fill in a reasonable amount of the bed, but w/o understanding your config it's again, difficult to guess what's going wrong (and so we usually won't as we've found it to historically be a waste of time).  I suggest you leave this open until you can submit your configs, or close and open a new issue when you have them.

f4bul1z3r, 2017-09-15:
Even if my assumption is incorrect the only logical way to level the bed is to probe it in some identifyable pattern. My printer seems to be randomly probing only about 1/4 of the bed. When I say random I mean that there is no logical pattern to the probing and not that it probes differently each time.

f4bul1z3r, 2017-09-15:
I will upload my config files when I get home

f4bul1z3r, 2017-09-15:
https://ghostbin.com/paste/379t8 - Configuration.h
https://ghostbin.com/paste/rq3u9 - Configuration_adv.h 

please only address the UBL random bed leveling issue in this thread please. I have addressed the compilation in issue in issue #7669

fiveangle, 2017-09-16:
If you disable the mesh validation feature (the code where sync_plan_postion_e() compile failure occurs), does it still limit the probing to one quadrant ?  Which quadrant is it liimited too ?

fiveangle, 2017-09-16:
Quadrant probe limit may be related to #7572

f4bul1z3r, 2017-09-16:
I cannot compile unless I disable that line with UBL enabled. 
Its the upper left quadrant.

I have tried using bilinear linear and 3 point and that all have bugs that cause terrible leveling on my delta. bilinear is the only one that is closest to working but it still does not level the bed correctly (one half of the print is being printed too high off the bed and the other is too close to the bed)

f4bul1z3r, 2017-09-16:
Also is there a slack group chat or something i can use since github issue messaging is kinda bad on my phone

Roxy-3D, 2017-09-16:
> When I say random I mean that there is no logical pattern to the probing and not that it probes differently each time.

UBL tries to probe points as close to the starting position of the nozzle as possible.   So, the nozzle will criss-cross across the 'known area' to the edge of the 'currently unknown' parameter.    This is done because it is thought that the nozzle will be started in a 'safe' location, and points close to the start should still be 'safe'.    If the nozzle is starting to get too close to the bed, it can be stopped.   But because the nozzle is working its way out from the starting position, it won't get in trouble without the user having a chance to stop it.

f4bul1z3r, 2017-09-16:
Whats funny is that I tried UBL on an older version and it probed in a weird way but it probed the whole bed. When it began printing I got the same issue as #7673 but instead of using bilinear I was obviously using UBL.

f4bul1z3r, 2017-09-16:
Ok @Roxy-3D your description matches what I saw with the older version but not what I saw with 1.1.5. With 1.1.5 there was movement similar to what you described but only staying in about the top left quadrant

Tannoo, 2017-09-17:
Probe boundries smaller than bed size with inset too much?

Tannoo, 2017-09-17:
Nevermind... deltas are different.

closed 2017-11-14 by thinkyhead