MarlinFirmware/Marlin #7670: Setting modifications via IDE 1.8.4 are not taken into account

Jacques-D, 2017-09-15:
Hello,

When I change motion settings (such as Z-offset or steps per unit) in configuration.h via Arduino IDE v1.8.4, and then upload the changes to the board, these changes are not taken into account. However, modifying the name of the printer did work when I modified that via Arduino IDE.

I did not get any error during compiling, and the upload of the firmware modifications do not raise any error either.
I've only tried to modify the printer name, the Z-offset and the steps per unit.

I'm guessing it is only a problem on my side because such bug would have been noticed by now. However, I don't know how to solve it, and it is a bit frustrating...
Can it be link to some EEPROM setting that I messed up ?

Note that if I change the step per unit using the M92 command, it does work.

Thank you,
Jacques

KevMags, 2017-09-15:
EEPROM values override firmware unless uploading a different version.  Erase EEPROM before uploading firmware, restore failsafe from LCD or send M502 to reset to firmware values.

Jacques-D, 2017-09-18:
Thank you for your feedback! I thought that uploading a new firmware through Arduino IDE was erasing everything.
I will try what you suggested tonight if I can and let you know!
Thank you!

Tannoo, 2017-09-19:
In fact, it does not. I've been bitten by that before.

pmbroth, 2017-09-21:
I have tried to initialize the eprom,  send M502, but upon uploading variables are not taken such as the Zoffset.  Any Ideas?  I'm using the Arduino 1.8.4   MKS GEN 1.4

pmbroth, 2017-09-21:
After the third try, it finally took...

closed 2017-11-27 by thinkyhead