MarlinFirmware/Marlin #7673: Bilinear bed leveling does not level the bed properly

f4bul1z3r, 2017-09-16:
### Bug Report
- Description: Bed is not leveled properly after BLTouch bilinear bed leveling
- Expected behaviour: I expect Marlin to take into account differences in height that are caused by my bed not being perfectly level and therefore creating a first layer that is consistent in height.
- Actual behaviour: If my circular bed was a square where 0,0 is the bottom left and 10,10 if the top right. when printing on the half closest to 0,0 (bed is halved diagonally) the nozzle prints too close to the bed and when printing on the other half closest to 10,10 the nozzle prints too far up from the bed. 
- Steps to reproduce: run a G28 than a g29 and commence a print that does NOT execute a G28 again.
im using the latest 1.1.5

f4bul1z3r, 2017-09-16:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1308641/Configuration.zip)

f4bul1z3r, 2017-09-16:
That should include both configuration files

guiseco, 2017-09-18:
Same problem here !

pandel, 2017-09-21:
Just for the record, I have the same problem. And as I wasn't able to solve it, I returned to linear leveling. That works. If I now print some kind of checkerboard pattern, you can clearly see, that my board is a bit bowed at some points, but it isn't completely lowered on the left side and raised on the right side at all.

As I read this error description more than once, perhaps there is something in the bilinear leveling algorithm, what is causing this to happen?

guiseco, 2017-09-21:
Mine is raised on the left side and lowered on the right side.

infernocs, 2017-12-15:
Facebook is full of people with the same problem, left goes high and right side low.

benlye, 2017-12-15:
I'm something of a broken record on this, but it's because I've been through it myself.  I realize my opinion is not always what people want to hear, but I have never seen any evidence to prove it wrong:

**Bi-linear leveling works.  If you get a leveling effect which is too high on one side and too low on the other, your printer has a mechanical, electrical, or configuration problem.**

List of things to check (based on how I and others have resolved this issue):
* Is your sensor accurate?  Is it accurate at all points on the bed (especially near the edges)?
* Are you giving your probe the voltage it requires?
* Is your sensor's X and Y offset config accurate?  It *must* be.  Don't rely on offsets provided by anyone else for your printer/sensor/mount - measure it yourself!
* Are your rods/rails straight?
* Does your wiring harness pull on the print head as it traverses across the bed?
* Have you tried probing with the heaters off to eliminate electrical interference from the heaters?
* Have you tried separating/shielding your probe wiring from any high-current wires in the wiring loom?

I spent days troubleshooting this problem on my printer, spending hours trying to 'debug' the bi-linear code, only to find that the problem was solved when I correctly calibrated the Y offset of my leveling sensor - the Y offset was out by just 2mm, which was the source of my issue (see #5898 for the full story).  Bi-linear leveling now works perfectly for me.

Bottom line: The problem is most likely your printer, not Marlin.  Same goes for the people on Facebook.  Rather than waiting for someone to fix the software, spend time checking (and double-checking) the list above.

pandel, 2017-12-15:
@benlye 
I have to say that in september, when I first tried bilinear leveling during August this year with the bugfix-1.1.x available at this time, I had the exact problem. As I wasn't able to fix it and because linear lev. worked fine so far, I never tried again. Now some days ago I had to reflash Marlin because of an extruder change, I tried bilinear again and now it works! I don't know why, I didn't change anything mechanically, sensor offset is the same, whatever... But, it works and that is everything that counts ;-)

benlye, 2017-12-15:
> ...because of an extruder change...

> I didn't change anything mechanically

What was the extruder change?  Perhaps it was enough to correct a latent geometry problem?  Anyway, glad it's working for you now :-)

pandel, 2017-12-15:
No no ;-) I simply swapped my bowden geared extruder with another bowden geared extruder :D - same position, same tube length, same cable position...

closed 2017-12-15 by thinkyhead