MarlinFirmware/Marlin #7674: Bug with PROBE_MANUALLY, LCD_LEVELING and DELTA with BILINEAR ABL

MagoKimbra, 2017-09-17:
------------------------------------------------------------
### Bug Report
I apologize for my bad English.
Then when the ABL bilinear routine is launched by LCD, the check of points is done by manual_probe_index, but in routine G29 some points are skipped when they are not on the plate. It happens that the G29 routine ends before the routine in the LCD that continues to call G29 V1 by resuming the G29 main process. I tried this with a Ramps Card a 3D Gadget screen and basic Kossel mini delta settings. Connecting the host to the board and starting the calibration procedure from display, the grid set to 5 x 5. The display marks 1/25 but already host tells me point 2 of 25 because the first is out of range. You get about 20/25 on the display when the host exits the grid and ABL is done, but the display continues to ask for points to be picked up to 25/25 by sending the G29 V1 command again in enquee and restarts the procedure. When the display finishes says to save in eeprom, but the printer stops waiting for the new points.
I tried to modify the display code by trying to synchronize the two things, but you have to deal with MBL and UBL that you have in Marlin ...
Thanks again and sorry ...

thinkyhead, 2018-02-09:
Still needs a patch. Will work on it.

boelle, 2019-02-19:
@thinkyhead has this been patched since?

boelle, 2019-03-06:
@thinkyhead ^^^^

boelle, 2019-03-16:
works fine for me

```
Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 +0.430 +0.650 +0.628
Recv:  1 +0.580 +0.859 +1.013
Recv:  2 +0.671 +0.821 +1.010
Recv: 
Recv: X:164.00 Y:-8.00 Z:5.26 E:0.00 Count X:34240 Y:30720 Z:4979
```

@MagoKimbra tried this out since you created the issue?

MagoKimbra, 2019-03-17:
Bolle in september 2017!! Now i think the problem not exist... Sorry if i do not closed...

boelle, 2019-03-17:
well everyone should have a fair chance to get back to us :-D

but it's up to you to close, you should be able to see a close button below, but only you can see it since you created issue :-D

closed 2019-03-17 by MagoKimbra